srware/meta-iot-cloud #3: aws-iot-device-sdk-embedded-C version?

brwhaley, 2016-12-28:
Are there plans to update meta-iot-cloud to use the 2.x version of the AWS embedded c sdk?  Thanks.

srware, 2016-12-29:
Hi,

Amazon stopped putting out OpenSSL builds of the embedded C SDK starting with version 2.x deciding to make it more focused on low power devices (mbed etc...). For platforms running full-fat Linux with OpenSSL or equivalent there is now a C++ SDK: https://github.com/aws/aws-iot-device-sdk-cpp

I will be adding support for the new C++ SDK shortly but the embedded C SDK will likely be staying at 1.x unless there is a specific need to port 2.x to OpenSSL which isn't satisfied by the C++ SDK..

Scott

closed 2017-01-04 by srware

brwhaley, 2017-01-06:
I wasn't aware of the C++ SDK but will take a look.  Thank you!

Ben