srware/meta-iot-cloud #4: aws-iot-device-sdk-cpp: do_package_qa errors

brwhaley, 2017-01-08:
Hello,

Thank you for adding the new recipe!  I'm able to compile, but am getting several errors during the package qa step of this nature:

QA Issue: non debug package contains .debug directory

Any help would be appreciated.

Thanks,
Ben

scooter1556, 2017-01-09:
Hi Ben,

Thanks for reporting. I was building on newer versions which are smart enough to automatically add the debug folders to the debug package. I've just noticed on 2.0 (Jethro) it doesn't and throws a bunch of errors.

I will update the recipe tomorrow to get that fixed.

Scott

brwhaley, 2017-01-09:
Thanks for all your help!

Ben

srware, 2017-01-10:
Hi Ben,

No problem. The issue should now be fixed :)

I haven't had time to test the C++ SDK extensively so feel free to contact me if there are any issues with the libs/binaries the recipe creates.

Scott

closed 2017-01-10 by srware