srware/meta-iot-cloud #6: bitbake dependencies error

asebak, 2017-01-14:
I am trying to build the azure-c-shared-utility and I am getting the following error: 

ERROR: Required build target 'azure-c-shared-utility' has no buildable providers.
Missing or unbuildable dependency chain was: ['azure-c-shared-utility', 'glibc-dev']

srware, 2017-01-14:
Which build system/toolchain (1.8, 2.0, 2.2, Wind River???). If you can give me some more information about how you are building the packages I can try to replicate and fix the issue.

Regards.

asebak, 2017-01-14:
i am just building it directly "bitbake azure-c-shared-utility".  I am not usinig an intel bsp though.  I am using my own.  I don't know if this is a generic yocto layer for all boards or specifically for intel boards?

srware, 2017-01-15:
The layer should be fairly generic but has only been tested with Poky based build environments. Glibc is included with OpenEmbedded Core so it's rare that this recipe isn't available which is why I asked about your build environment. Glibc-dev is a hard-coded recommended dependency for the 'azure-c-shared-utility' recipe to get around some bugs but it seems your build system doesn't have any recipes which provide it. I will see if I can find a better workaround than hard-coding to see if this resolves the issue.

asebak, 2017-01-15:
Just to give you an update when I commented out those dependencies I managed to successfully create the iothub_client.so file.  But when I use it on my device it gives an issue about Cannot resolve symbol in the shared object file...  I noticed there's a flag -Duse_installed_dependencies:bool={ON/OFF} not sure if that has to do with that.  Thanks for looking into it.

srware, 2017-03-21:
Hi @asebak,

Is this still an issue with the latest master branch?

Thanks

srware, 2017-03-25:
Closing this issue for now as it should be solved. Feel free to re-open if it persists.

closed 2017-03-25 by srware