srware/meta-iot-cloud #9: build azure-iot-sdk error on pulsar linux 8

aaronadlink, 2017-07-21:
I got pulsar linux 8
https://github.com/WindRiver-OpenSourceLabs/wr-core.git
Added meta-iot-cloud to pulsar linux 8 and build azure-iot-sdk error

Error message :  
aaron@ubuntu:~/pulsar_n/wr-core/build-intel-apollolake$. init-intel-apollolake-env
aaron@ubuntu:~/pulsar_n/wr-core/build-intel-apollolake$ bitbake packagegroup-cloud-azure
Loading cache: 100% |#######################################################################################| ETA:  00:00:00
Loaded 6985 entries from dependency cache.
Parsing recipes: 100% |#####################################################################################| Time: 00:00:02
Parsing of 3071 .bb files complete (3068 cached, 3 parsed). 6989 targets, 587 skipped, 9 masked, 0 errors.
WARNING: No recipes available for:
  /home/aaron/pulsar_n/wr-core/layers/meta-secure-env/recipes-support/ima-evm-utils/ima-evm-utils_git.bbappend
  /home/aaron/pulsar_n/wr-core/layers/meta-virtualization/recipes-kernel/linux/linux-yocto_4.8.bbappend
  /home/aaron/pulsar_n/wr-core/layers/meta-gateway/recipes-core-luci/busybox/busybox_1.22.1.bbappend
  /home/aaron/pulsar_n/wr-core/layers/wrlabs-integration/recipes-daemons/tftp-hpa/tftp-hpd_5.2.bbappend
NOTE: Resolving any missing task queue dependencies
ERROR: Nothing RPROVIDES 'boost-python' (but /home/aaron/pulsar_n/wr-core/layers/meta-iot-cloud/recipes-azure/azure-iot-sdk/azure-iot-sdk_1.1.13.bb RDEPENDS on or otherwise requires it)
NOTE: Runtime target 'boost-python' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['boost-python']
NOTE: Runtime target 'azure-iot-edge' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['azure-iot-edge', 'azure-iot-sdk', 'boost-python']
ERROR: Required build target 'packagegroup-cloud-azure' has no buildable providers.
Missing or unbuildable dependency chain was: ['packagegroup-cloud-azure', 'azure-iot-edge', 'azure-iot-sdk', 'boost-python']

Summary: There was 1 WARNING message shown.
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.

srware, 2017-07-21:
Hi @aaronadlink ,

The issue here is that the Python bindings for Boost are not built by the Boost recipe by default so you have to manually specify the PACKAGECONFIG settings in your build config.

Add the following line to your 'conf/local.conf' file or create a 'conf/auto.conf' file with the following contents:

`PACKAGECONFIG_pn-boost = "python"`

That should solve it.

aaronadlink, 2017-07-21:
it is working now
thanks.

closed 2017-07-21 by aaronadlink