srware/meta-iot-cloud #10: Python install path incorrect in azure-iot-sdk bitbake recipe

rk-bp, 2017-08-04:
In the ob_libinstall lines below "${B}" should be "${B}/c".

do_install_append() {
	# Python
	if ${@bb.utils.contains('PACKAGECONFIG','python','true','false',d)}; then
		install -d ${D}${PYTHON_SITEPACKAGES_DIR}
		oe_libinstall -C ${B}/python/src -so iothub_client ${D}${PYTHON_SITEPACKAGES_DIR}
		oe_libinstall -C ${B}/python_service_client/src -so iothub_service_client ${D}${PYTHON_SITEPACKAGES_DIR}
	fi
}

srware, 2017-08-04:
What issue are you seeing? 

The build directory ${B} isn't the same as the git source directory ${S} so as far as I can see the current paths are correct and the Python libs are being packaged as expected. '${B}/c' doesn't exist...

rk-bp, 2017-08-04:
I am seeing this when I pulled the meta-iot-devkit layer into Yocto Daisy.  The path, at least in Yocto Daisy, is ${B}/c/python/src rather than ${B}/python/src.  If it is not modified in the reciped, Yocto throws a "path not found" error.

closed 2017-08-04 by rk-bp

srware, 2017-08-04:
Interesting... I haven't tested on 1.6 as it's pretty old now. The layer is targeted at 2.0 and above. Something must be different in the CMake class or version of CMake.

I could make ${B} '?=' so you can override in a bbappend for your build?

rk-bp, 2017-08-04:
Also, the generated package could not be installed on Daisy even with the change I made to the recipe to the path IRT Python because the package generated by the bitbake recipe generates ${PN}-c as the package name while the Yocto Daisy installer is looking for ${PN} package name.  We are not upgrading to Jethro until second quarter next year.  I will see what I can do with a .bbappend.

rk-bp, 2017-08-04:
We are doing a POC, so I hacked your .bb (attached) to get the SDK to build in Yocto Daisy.  I will now revert these changes and work on a bbappend.

[azure-iot-sdk_1.1.13.bb.zip](https://github.com/intel-iot-devkit/meta-iot-cloud/files/1201247/azure-iot-sdk_1.1.13.bb.zip)

scooter1556, 2017-08-04:
I've made some changed to the recipe which will hopefully help. You should just need to put the following in a .bbappend.

`B = ${WORKDIR}/build/c`

rk-bp, 2017-08-07:
Thanks.  I will create a .bbapend and try today.

rk-bp, 2017-08-08:
[azure-iot-sdk.zip](https://github.com/intel-iot-devkit/meta-iot-cloud/files/1206738/azure-iot-sdk.zip)
I had to modify your recipe slightly (attached) to get a successful Daisy build

scooter1556, 2017-08-08:
I can see your .bbappend but I guess you didn't mean to include the '_java' recipe?

rk-bp, 2017-08-08:
[azure-iot-sdk.zip](https://github.com/intel-iot-devkit/meta-iot-cloud/files/1208244/azure-iot-sdk.zip)

srware, 2017-08-08:
Thanks for re-uploading.

If you pull the latest changes to the layer does the bbappend content I posted previously work?

`B = "${WORKDIR}/build/c"`

You should be able to override 'B' in your bbappend rather than adding another variable 'C'.

Thanks.

rk-bp, 2017-08-08:
Using an overridable B in the recipe results in an extra "/c" within the installation path.

rk-bp, 2017-08-08:
It is really quite strange...

srware, 2017-08-08:
Actually, that's my bad...

'B' is used in the CMake class to determine where to put build files so it will automatically make the directory specified in that variable.

I've just pushed another commit which will hopefully solve things with the following in a bbappend:

`OUTDIR = "${WORKDIR}/build/c"`

rk-bp, 2017-08-08:
Thanks, I will test it right now.

rk-bp, 2017-08-08:
Works.  Thanks.  ;-)