srware/meta-iot-cloud #15: gRPC recipe does not work for C++

xinxi531, 2017-09-13:
This gRPC recipe can only build grpc library for C support but not C++.

srware, 2017-09-15:
Hi @xinxi531 ,

I'm not entirely sure why the gRPC recipe is still here actually, it's not used by any other recipes as far as I know. I can remember originally writing the recipe for this library and the 'Makefile' is a mess. It overrides so many standard variables set by the Yocto build system that cross compiling becomes a major headache. It also uses 'protoc' to build all plugins (including c++ I believe) which the Makefile can never find in sysroot because of the overrides so it fails miserably.

I'll happily accept a pull request for this but I don't have time to look into this further and create the Makefile patches necessary for this to build well with Yocto. If I get nothing I will probably just remove this recipe.

Thanks

xinxi531, 2017-09-15:
Hi Scott,

Thank you for your reply. Indeed I have tried to write a recipe to build gRPC++ but their Makefile for grpc++ library is just so messy. There are a lot of stuff going on and a lot of variables need to be set (some even need to be set twice). I am still working on it.

My suggestion is do not delete the recipe. It supports C and Python anyways.

I will let you know if there are any updates!

Thanks!

Xi.

srware, 2018-07-25:
Closing this issue now as the recipe has been removed from the layer given that there is now a gRPC recipe maintained in meta-oe.

closed 2018-07-25 by srware