srware/meta-iot-cloud #19: azure-iot-sdk build error

jefdriesen, 2018-01-26:
Hi,

The azure-iot-sdk recipe contains this code:
```
do_configure_prepend() {
	cd ${S}
	git config --global http.sslverify "false"
	git submodule update --init --recursive
}
```
But this breaks the build on our build server, because there is no `$HOME` variable set and hence the `git config' fails with this error:
```
DEBUG: Executing shell function do_configure
fatal: $HOME not set
WARNING: /home/bob/yocto/build/tmp/work/armv7ahf-neon-poky-linux-gnueabi/azure-iot-sdk/1.1.23-r0/temp/run.do_configure.2839:1 exit 128 from 'git config --global http.sslverify "false"'
ERROR: Function failed: do_configure (log file is located at /home/bob/yocto/build/tmp/work/armv7ahf-neon-poky-linux-gnueabi/azure-iot-sdk/1.1.23-r0/temp/log.do_configure.2839)
```
This can be fixed by removing the `--global` option. Using this option is a very bad idea anyway, because now you suddenly disable verification of SSL certificates for ALL git repositories! I wonder why the `http.sslverify` options needs to be disabled in the first place?

Jef

srware, 2018-01-26:
Thank you for flagging this. This hack was always on my ToDo list to fix.

The issue is a leftover fix for legacy Yocto versions which for some reason failed to execute git commands specified manually in the recipe with SSL enabled. This was a dirty hack to make it work but as you have realised also introduces other issues and isn't exactly good practice!

I will work on a fix and get it pushed hopefully today :)

jefdriesen, 2018-01-26:
Removing the `--global` option already fixes the build problem for me. But I also tested without the whole `git config` line, and that works just fine too! So that's probably a better solution.

I can create a pull request if that's easier for you.

srware, 2018-01-26:
I just tested with Yocto 2.3 and the default gitsm fetcher now appears to do a recursive pull by default so the configure prepend isn't required at all.

I need to check how far back the Yocto branches this is the case but for master I'm happy to remove the function completely. I will do this later today if you agree.

jefdriesen, 2018-01-26:
Sure, go ahead!

srware, 2018-01-26:
Done.

Let me know if you still have issues.

closed 2018-01-26 by srware