srware/meta-iot-cloud #20: azure-c-shared-utility build error

jefdriesen, 2018-02-14:
Commit 23b2acb4f54f7331586c14cc432cb82cd0eaccf1 seems to break the build, with this error:
```
ERROR: azure-c-shared-utility-1.0.52-r0 do_patch: Command Error: 'quilt --quiltrc /home/jdi/develop/yocto/build/tmp/work/armv7ahf-neon-poky-linux-gnueabi/azure-c-shared-utility/1.0.52-r0/recipe-sysroot-native/etc/quiltrc push' exited with 0  Output:
Applying patch Use-pkg-config-to-find-libs.patch
patching file CMakeLists.txt
Hunk #1 succeeded at 125 with fuzz 1 (offset -2 lines).
Hunk #2 FAILED at 420.
Hunk #3 succeeded at 427 (offset -29 lines).
1 out of 3 hunks FAILED -- rejects in file CMakeLists.txt
Patch Use-pkg-config-to-find-libs.patch does not apply (enforce with -f)
ERROR: azure-c-shared-utility-1.0.52-r0 do_patch: Function failed: patch_do_patch
ERROR: Logfile of failure stored in: /home/jdi/develop/yocto/build/tmp/work/armv7ahf-neon-poky-linux-gnueabi/azure-c-shared-utility/1.0.52-r0/temp/log.do_patch.24388
ERROR: Task (/home/jdi/develop/yocto/meta-iot-cloud/recipes-azure/azure-c-shared-utility/azure-c-shared-utility_1.0.52.bb:do_patch) failed with exit code '1'
```
So it looks like the patch doesn't apply correctly. The strange thing is that if I apply the patch manually using `git apply` or `patch -p1`, then there is no error:
```
$ git apply Use-pkg-config-to-find-libs.patch
../Use-pkg-config-to-find-libs.patch:26: trailing whitespace.
    
../Use-pkg-config-to-find-libs.patch:60: trailing whitespace.
        
warning: 2 lines applied after fixing whitespace errors.

$ patch -p1 < ../Use-pkg-config-to-find-libs.patch 
patching file CMakeLists.txt
```

srware, 2018-02-14:
Hi @jefdriesen ,

Which Yocto version are you based on so I can check it out? I tested on 2.4 as this is 'master' but am yet to check the patch with other branches.

Thanks

jefdriesen, 2018-02-14:
I'm using Yocto pyro (2.3).

srware, 2018-02-14:
Thanks for the info @jefdriesen 

I'm struggling to reproduce. I just did a clean build using pyro branches and master branch of this repo and the package built without issue. I have just pushed a change to make sure the recipe commit doesn't wonder. Can you do a 'cleanall' on this package and try again?

Thanks

jefdriesen, 2018-02-14:
The problem remains afer doing a cleanall.

srware, 2018-02-14:
Is that with the latest changes pulled from master?

Could you also try the 'pyro' branch. I pushed a back-ported patch to that branch for the older version so it would be interesting to know if your build environment fails with that also.

Thanks

jefdriesen, 2018-02-15:
There is something really strange happening on my side. If I clean the package, and try to build it again, then it looks like yocto did checkout the wrong commit!

It does checkout commit b6ce97005ebe01b8656a505533d1501295e1e928 instead of the commit that was specified in the `SRCREV` variable (currently 4fe03ef78499f8acb1b2f8526b6cea8c432140d8). In the attached [do_fetch.log](https://github.com/intel-iot-devkit/meta-iot-cloud/files/1727256/do_fetch.log) you can clearly see this wrong commit. I have no idea how I end up with this wrong commit.

But of course that explains why the patch doesn't apply!

srware, 2018-02-15:
@jefdriesen that's a new one on me...

I just checked my fetch log and it's basically the same but pulling the correct commit.

Maybe try getting rid of your 'tmp/cache' directory and try again? 

The only other thing which has changed in the recipe is the release branch is specified rather than just pointing to a commit on master but I see no reason why this should cause an issue...

jefdriesen, 2018-02-21:
I found the problem. We have another layer that had a bbappend for azure-c-shared-utility, where the SRCREV was set to b6ce97005ebe01b8656a505533d1501295e1e928. Sorry for the noise!

closed 2018-02-21 by jefdriesen

srware, 2018-02-21:
No problem. Glad you found the issue.