srware/meta-iot-cloud #22: question on including azure's certs to yocto build

JoeyR, 2018-03-15:
I was doing some digging on an issue we have and came across [#175](https://github.com/Azure/azure-iot-sdk-c/issues/175)

The solution for us would be to set the "TrustedCerts" option in the IoT Client connection, but that depends on us including these files into the image/sdk. 

For reference here are the [certs](https://github.com/Azure/azure-iot-sdk-c/tree/5ff695a91c95b0251f69bbb0b411d71b43fd2ff9/certs) we need to add.

Any help is appreciated!
Joey

srware, 2018-03-19:
Hi @JoeyR,

I will look into getting the certs installed alongside the SDK and get back to you.

Thanks.

srware, 2018-03-26:
Hi @JoeyR,

Looking into this some more it looks like the certs are designed to be compiled into your code (or the sample) rather than included on the filesystem in a standard format (e.g. pem). The certs.h file isn't used for the actual SDK libs so including just the header file with the dev package would be pointless. I assume the expected behaviour is for you to include the cert.c and certs.h alongside your own code and compile the certs into the resultant binary.

Thanks.

JoeyR, 2018-03-26:
@srware 

Thanks for the update.

I was mainly looking at this problem from an update/maintenance perspective by including them in our toolchain. Thanks for your help and I'll build them in!

Joey

closed 2018-03-26 by JoeyR