srware/meta-iot-cloud #32: How to update azure-iot-sdk-c to version 1.2.4?

JoeyR, 2018-05-16:
I'm not a bitbake master so I was wondering if those repos could get updated, also if it's an easy thing to do I would love to learn how to do it!

srware, 2018-05-17:
Hi @JoeyR ,

The main hold-up getting this recipe updated is that it is a dependency for the "Azure IoT Edge" recipe and this isn't getting updated much these days. I think in light of the fact v1 of Azure IoT Edge is now deprecated I will remove this from the 'master' branch which will allow me to update the C SDK recipe without breaking things. I can look into this later today.

In terms of updating it largely depends on how drastically the SDK has changed. Sometimes updating the recipe is as easy as updating the git commit to the desired release and potentially changing license checksums if this has been updated. If there are bigger changes such as new libs, additional dependencies or compile flags then more changes might be required such as updating the list of files to be installed in the image.

Thanks.

JoeyR, 2018-05-17:
Thanks for the info!

I have no issue with git on version 1.0.32, but I'm getting an error by changing the srcrev (and bb names) to the 1.2.4 commit:

`WARNING: azure-iot-sdk-c-1.2.4-r0 do_fetch: Failed to fetch URL gitsm://github.com/Azure/azure-iot-sdk-c.git, attempting MIRRORS if available

ERROR: azure-iot-sdk-c-1.2.4-r0 do_fetch: Fetcher failure: Unable to find revision cf2b33b6888c2322647737852499a0d7e80fcde7 in branch master even from upstream     
ERROR: azure-iot-sdk-c-1.2.4-r0 do_fetch: Fetcher failure for URL: 'gitsm://github.com/Azure/azure-iot-sdk-c.git'. Unable to fetch URL from any source.
ERROR: azure-iot-sdk-c-1.2.4-r0 do_fetch: Function failed: base_do_fetch
ERROR: Logfile of failure stored in: /home/build/patching/mlinux4.0.0dev11/build/tmp/work/arm926ejste-mlinux-linux-gnueabi/azure-iot-sdk-c/1.2.4-r0/temp/log.do_fetch.8541
ERROR: Task (/home/build/patching/mlinux4.0.0dev11/layers/user-layer/recipes-bluepillar/azure-iot-sdk-c
/azure-iot-sdk-c_1.2.4.bb:do_fetch) failed with exit code '1' `

srware, 2018-05-17:
This is because the commit you are fetching isn't on the 'master' branch so you would need to specify this in "SRC_URI". Try this commit: ad28b943aceea95df07d27d97206c987ebdd86f8

Also bear in mind that the SDK depends on a few other libraries which will also likely require updating to compile successfully (azure-c-shared-utility etc...).

srware, 2018-06-01:
Recipe has been updated in master and sumo branches.

Thanks.

closed 2018-06-01 by srware