srware/meta-iot-cloud #35: python-knack: License checksum is invalid

derekstraka, 2018-07-24:
```
ERROR: python-knack-0.4.1-r0 do_fetch: Checksum failure fetching https://github.com/Microsoft/knack/raw/0.4.1/LICENSE;name=license
ERROR: python-knack-0.4.1-r0 do_fetch: Fetcher failure for URL: 'https://github.com/Microsoft/knack/raw/0.4.1/LICENSE;name=license'. Checksum mismatch!
File: '/var/build/build/downloads/LICENSE' has md5 checksum 650b869bd8ff2aed59c62bad2a22a821 when bbfbc44677c93751d972e8b36751a695 was expected
File: '/var/build/build/downloads/LICENSE' has sha256 checksum 7c34d28e784b202aa4998f477fd0aa9773146952d7f6fa5971369fcdda59cf48 when b9e1c9b3b348a58186dd4fc69c994cf26c0ded0d91ebe4bd201e6f7d05ead8ec was expected
If this change is expected (e.g. you have upgraded to a new version without updating the checksums) then you can use these lines within the recipe:
SRC_URI[license.md5sum] = "650b869bd8ff2aed59c62bad2a22a821"
SRC_URI[license.sha256sum] = "7c34d28e784b202aa4998f477fd0aa9773146952d7f6fa5971369fcdda59cf48"
Otherwise you should retry the download and/or check with upstream to determine if the file has become corrupted or otherwise unexpectedly modified.

ERROR: python-knack-0.4.1-r0 do_fetch: Function failed: base_do_fetch
ERROR: Logfile of failure stored in: /var/build/build/tmp/work/core2-64-poky-linux/python-knack/0.4.1-r0/temp/log.do_fetch.141
ERROR: Task (/var/build/poky/meta-iot-cloud/recipes-devtools/python/python-knack_0.4.1.bb:do_fetch) failed with exit code '1'
```

I can update the checksum, but it seems odd that the checksum differs

derekstraka, 2018-07-24:
@srware - I will do the fixes for this one and #34; I just want to know what you'd prefer as an action

srware, 2018-07-25:
Thanks for reporting @derekstraka . I agree this is bad practice, this one fell through the net. I usually use ${PN}-${PV}-license when downloading license files separate from the main package. Even better would be if these Python packages included the license file...

This is now fixed in master.

commit 8c79cef5 (2018-07-25): Python Knack: Fix license issue (Fixes #35)
Python Azure MultiAPI Storage: Fix license issue (Fixes #34)
closed 2018-07-25 by srware