srware/meta-iot-cloud #36: packagegroup-cloud-azure is not installable

derekstraka, 2018-07-25:
When using the following image, the package group `packagegroup-cloud-azure` is not installable

```
SUMMARY = "A small image just capable of allowing a device to boot with azure support"
CORE_IMAGE_EXTRA_INSTALL += "packagegroup-cloud-azure"
IMAGE_INSTALL = "packagegroup-core-boot ${CORE_IMAGE_EXTRA_INSTALL}"
IMAGE_LINGUAS = " "
LICENSE = "MIT"
inherit core-image
IMAGE_ROOTFS_SIZE ?= "8192"
IMAGE_ROOTFS_EXTRA_SPACE_append = "${@bb.utils.contains("DISTRO_FEATURES", "systemd", " + 4096", "" ,d)}"
```

Results in 
```
ERROR: core-image-minimal-azure-1.0-r0 do_rootfs: Could not invoke dnf. Command '/var/build/build/tmp/work/qemux86_64-poky-linux/core-image-minimal-azure/1.0-r0/recipe-sysroot-native/usr/bin/dnf -v --rpmverbosity=debug -y -c /var/build/build/tmp/work/qemux86_64-poky-linux/core-image-minimal-azure/1.0-r0/rootfs/etc/dnf/dnf.conf --setopt=reposdir=/var/build/build/tmp/work/qemux86_64-poky-linux/core-image-minimal-azure/1.0-r0/rootfs/etc/yum.repos.d --repofrompath=oe-repo,/var/build/build/tmp/work/qemux86_64-poky-linux/core-image-minimal-azure/1.0-r0/oe-rootfs-repo --installroot=/var/build/build/tmp/work/qemux86_64-poky-linux/core-image-minimal-azure/1.0-r0/rootfs --setopt=logdir=/var/build/build/tmp/work/qemux86_64-poky-linux/core-image-minimal-azure/1.0-r0/temp --nogpgcheck install packagegroup-core-boot packagegroup-cloud-azure run-postinsts' returned 1:
DNF version: 2.7.5
cachedir: /var/build/build/tmp/work/qemux86_64-poky-linux/core-image-minimal-azure/1.0-r0/rootfs/var/cache/dnf
Added oe-repo repo from /var/build/build/tmp/work/qemux86_64-poky-linux/core-image-minimal-azure/1.0-r0/oe-rootfs-repo
repo: using cache for: oe-repo
not found deltainfo for: oe-repo
not found updateinfo for: oe-repo
oe-repo: using metadata from Wed 25 Jul 2018 10:54:09 AM UTC.
Last metadata expiration check: 0:00:01 ago on Wed 25 Jul 2018 10:54:10 AM UTC.
--> Starting dependency resolution
--> Finished dependency resolution
Error: 
 Problem: package python-azure-iothub-service-client-1.2.2-r0.core2_64 conflicts with python-azure-iot-sdk provided by python-azure-iothub-device-client-1.2.2-r0.core2_64
  - package packagegroup-cloud-azure-0.16-r0.noarch requires python-azure-iothub-service-client, but none of the providers can be installed
  - package packagegroup-cloud-azure-0.16-r0.noarch requires python-azure-iothub-device-client, but none of the providers can be installed
  - conflicting requests
(try to add '--allowerasing' to command line to replace conflicting packages or '--skip-broken' to skip uninstallable packages)

ERROR: core-image-minimal-azure-1.0-r0 do_rootfs: Function failed: do_rootfs
ERROR: Logfile of failure stored in: /var/build/build/tmp/work/qemux86_64-poky-linux/core-image-minimal-azure/1.0-r0/temp/log.do_rootfs.7421
ERROR: Task (/var/build/poky/meta-iot-cloud-test/recipes-core/images/core-image-minimal-azure.bb:do_rootfs) failed with exit code '1'
```

srware, 2018-07-25:
Thanks for reporting @derekstraka.

This should now be fixed.

closed 2018-07-25 by srware

derekstraka, 2018-07-25:
Verified.  Thanks!

commit abd54016 (2018-07-25): Azure: Fix Azure package installation (Fixes #36)
commit 9694d51e (2018-07-26): Azure: Fix Azure package installation (Fixes #36)
commit 24d98433 (2018-07-27): Azure: Fix Azure package installation (Fixes #36)