srware/meta-iot-cloud #42: Any plans updating node-red recipe to newer version?

jostor, 2018-12-05:
Are there any plans updating node-red recipe to a newer version soon? I tried doing it myself, by just renaming the file to node-red_0.19.5.bb, and updating the checksums. This worked fine when I was using the sumo branch, but on thud it isn't working. I get "ERROR: Function failed: sysroot_strip". See full log here: https://gist.github.com/jostor/d0318ee399ced56b0ae168ae8a72d1a7

srware, 2018-12-05:
Hi @jostor,

I've been updating recipes on request so I'm happy to update the Node-RED recipe in master. I'd be reluctant to update in any of the stable branches but I can make sure it is backwards compatible so you can use it with the "thud" branch if you wish.

I will try and get around to it later today.

Thanks.

srware, 2018-12-05:
@jostor, I have pushed a 0.19.5 recipe to master and tested with the 'thud' branch also.

Let me know if you have any issues.

closed 2018-12-05 by srware