srware/meta-iot-cloud #43: error while compiling samples of azure-iot-sak-c in rocko branch

vsivanagulu, 2019-01-08:


srware, 2019-01-08:
Hi @vsivanagulu,

This is semi-expected because the samples don't support building against shared libraries so the samples are linking against libs which don't exist. It's a bug with the SDK build system where the flag "build_as_dynamic" isn't taken into account for samples. I never bothered to patch it for the recipe because most of the samples require code changes like setting of the connection string to be useful. Compiling them as part of the SDK recipe is of very limited use.

I may add an example recipe showing how one of the samples could be compiled in the correct manner separate from the SDK build. Hopefully I will get time in the near future to do so.

vsivanagulu, 2019-01-09:
@srware  thats great work. Please post that example recipe. Also I have time now to do that Guide me to write that sample recipe. I would really very happy to work and support for "meta-iot-cloud" team.
Note: I have compiled libs succesfully How can I test azure-iot-sdk-c without samples. is there any other way.

srware, 2019-01-14:
I have added an example showing how to compile one of the Azure IoT SDK samples in it's own recipe over here: [meta-iot-cloud-examples](https://github.com/srware/meta-iot-cloud-examples)

The example code is taken straight from the SDK repository with a slightly modified cmake build script. Something similar should work to build your own code or other samples against the meta-iot-cloud recipes.

In terms of testing the idea here is that you would already have known working code which depends on the Azure IoT SDK. For instance, I create an application on my Ubuntu dev machine which requires the Azure IoT SDK, I then want to add this to my Yocto image so I create a recipe for my code which depends on azure-iot-sdk-c provided by this layer.

closed 2019-01-29 by srware