srware/meta-iot-cloud #46: Not able to see Sensors in Node RED

SatinderSidhu, 2019-04-06:
I am developer, got the intel grove kit from one of the session attended lately. I am have started a  lab at home but hitting a road blocker. I have connected everything, but not able to detect the sensors in dashboard, so can't run any program. Is there troubleshooting i can do?

SatinderSidhu, 2019-04-06:
[image]

[image]

SatinderSidhu, 2019-04-06:
[image]

srware, 2019-04-08:
Hi @SatinderSidhu,

This repository isn't connected to the developer kit you are inquiring about so I am going to close this.

You should ask the question in the Intel forums.

Thanks

closed 2019-04-08 by srware