srware/meta-iot-cloud #47: Use protobuf recipe from meta-oe

paul-betafive, 2019-04-11:
A recipe for protobuf is already provided in meta-oe. This is the version against which other layers have been tested. Once we add in meta-iot-cloud to our build, we see a failure to build the armnn package as it is no longer building against the expected protobuf version.

Is it possible to scrap the protobuf recipe in this layer and instead use the one from meta-oe? If not then the recipe in this layer should be renamed to something like `protobuf3.3` so that it doesnt' clash with the `protobuf` recipe used by other layers.

srware, 2019-04-11:
Hi @paul-betafive,

Good catch. This isn't needed in newer branches since python-protobuf was added to meta-python so I can remove it back as far as rocko (2.4). I'll get onto it later today/tomorrow.

For the other branches I will change the name of the recipe if required.

Thanks again.

paul-betafive, 2019-04-11:
rocko is the version I saw the issue on so personally I'm not interested in anything older. Currently trying to work around by setting `PREFERRED_VERSION_protobuf`. I'm happy to test any potential fix you have for the rocko branch though.

srware, 2019-04-11:
I have just pushed some changes to the rocko branch.

Let me know if this fixes the issue.

Thanks.

srware, 2019-04-16:
This should now be fixed so closing for now.

Feel free to reopen if there is still an issue.

closed 2019-04-16 by srware