srware/meta-iot-cloud #48: recipes-google-clous/python missing in latest versions

hicklin, 2019-06-21:
I noticed that the recipes for python APIs to the Google Cloud Platform are missing from the thud and warrior branches. Is this intentional?

srware, 2019-06-21:
Hi @hicklin,

It is intentional mainly because they were a nightmare to maintain and kept breaking.

If there is a need/want for one or more of them I will happily add them back so please let me know which ones you are using and I will see what I can do.

Thanks,

Scott

hicklin, 2019-06-21:
I feel your pain. I am trying to build them myself at the moment. The main recipe I currently need is pubsub. It seems that the recipes in the older versions are a bit outdated for my needs. 

I'm happy to help out with updating the recipes that I need.

srware, 2019-06-21:
@hicklin,

Pubsub shouldn't be too difficult to add back in to master. I will take a look later on today.

Thanks.

hicklin, 2019-06-21:
I am in the process of updating all the recipes that PubSub depends on to
the latest tagged versions in their respective repositories. I can push the
updates and you can review them if that saves you time.

srware, 2019-06-21:
Sounds good. Appreciate it :)

hicklin, 2019-06-21:
I have finished and tested all the recipes required to get PubSub working. I am working on a local feature brunch called thud-google-cloud-python. Can you give me access to commit this branch?

Note: I have not included the python.bb files. The reason being that I am not using python2 and will not be testing them. Also, some of the packages have published that support for python2 will be dropped from 1st of January 2020.

srware, 2019-06-21:
@hicklin,

Thanks for your efforts here. Normal procedure is to fork the repo and submit a PR against the feature branch in your fork. Unfortunately I can't give you write access to this repo.

I will still need to add python2 variants for any Python packages which support 2 and 3 for the next couple of Yocto releases I guess but I am happy to add these.

Thanks.

hicklin, 2019-06-21:
OK, I will leave it up to you to add and test the python2 variants. The PR has been sent. I am going to close the issue.

closed 2019-06-21 by hicklin