srware/meta-iot-cloud #51: Service file for Node-RED using the wrong path

jumo-sballing, 2019-08-23:
Since the YOCTO team has changed the install directory for node modules from $PREFIX/lib/node_modules to $PREFIX/lib/node, the [node-red.service](https://github.com/intel-iot-devkit/meta-iot-cloud/blob/master/recipes-node-red/node-red/node-red/node-red.service) file from your node-red [recipe](https://github.com/intel-iot-devkit/meta-iot-cloud/blob/master/recipes-node-red/node-red/node-red_0.20.5.bb) is broken.

You use `ExecStart=/bin/su root -c 'node /usr/lib/node_modules/node-red/red.js'`. Now it is `ExecStart=/bin/su root -c 'node /usr/lib/node/node-red/red.js'`.

This effects the branch warrior.

You can read the reasons for the change here:
https://git.yoctoproject.org/cgit/cgit.cgi/poky/commit/?id=2713d9bcc39c712ef34003ce8424416441be558e&showmsg=1

If you like, I can send you a PR.

srware, 2019-08-23:
Many thanks for the heads up. I will get it fixed :)

srware, 2019-08-23:
I just double-checked this and in the latest "master" and "warrior" Poky branches node modules are still installed to "/usr/lib/node_modules". The "/usr/lib/node" directory is just a symlink.

https://git.yoctoproject.org/cgit/cgit.cgi/poky/tree/meta/classes/npm.bbclass#n13

See this commit which reverted the change:

https://git.yoctoproject.org/cgit/cgit.cgi/poky/commit/meta/classes/npm.bbclass?id=a34d0d539e5fdf341541fb628652d22289e80512

In fact "thud" is the only branch affected by this and for that I will need to override the NPM_INSTALLDIR so that the symlinks created in "/usr/bin" point to the correct location...

Thanks.

srware, 2019-08-27:
No problem :)

closed 2019-08-27 by srware

jumo-sballing, 2019-08-27:
Thank you so much for checking this. 

That was my mistake. I just looked at our internal mirror of the Yocto repository and didn't see the revert commit in the official repo. I have now realized that everything is fine in the official Warrior Branch.

This can be closed now.

srware, 2019-08-28:
Let me patch that, it's on my list :)

Hopefully be fixed by EOD.

srware, 2019-08-28:
I just checked the 'thud' branch of this layer and it isn't using the npm.class for node-red which means it is installing to the correct location... There will be issues if attempting to use the 'master' branch of this layer with 'thud'. If this issue is relating to the 'thud' branch of this layer let me know.

srware, 2019-08-28:
Sure, I understand, however, I keep this layer in line with Yocto release branches so to ensure compatibility branches need to be the same for all layers. You can of course still use master and apply your own patches but I won't be able to patch 'master' to work with 'thud' upstream.

The fix for this should be pretty easy to add in a .bbappend though. Just add the following after 'inherit npm'

`NPM_INSTALLDIR = "${libdir}/node_modules/${NPMPN}"`

That should correct the install directory for 'thud'.

Thanks.