srware/meta-iot-cloud #58: dotnet_3.0.0.bb doesn't support arm32

trampster, 2019-10-20:
I added the following:
```
elif re.match('arm$', target_arch):
        d.setVar('SRC_FETCH_ID', '8ddb8193-f88c-4c4b-82a3-39fcced27e91/b8e0b9bf4cf77dff09ff86cc1a73960b')
        d.setVarFlag('SRC_URI', 'md5sum', '7c253a28becd85b9f19880d745e130ba')
        d.setVarFlag('SRC_URI', 'sha256sum', 'e0090dc52a1fbfcf7467fe874ddc58fb27a0c34fb4ab3c1124c476c92d553693')
        d.setVar('DOTNET_ARCH', 'arm')
```

which got me closer but then got this error:

ERROR: dotnet-3.0.0-r1 do_package_qa: QA Issue: /usr/share/dotnet/shared/Microsoft.NETCore.App/3.0.0/System.Net.Http.Native.so contained in package dotnet requires libcurl.so.4(CURL_OPENSSL_3), but no providers found in RDEPENDS_dotnet? [file-rdeps]
ERROR: dotnet-3.0.0-r1 do_package_qa: QA run found fatal errors. Please consider fixing them.
ERROR: dotnet-3.0.0-r1 do_package_qa: Function failed: do_package_qa
ERROR: Logfile of failure stored in: /chimera-src/build/tmp/work/armv7ahf-neon-tait-linux-gnueabi/dotnet/3.0.0-r1/temp/log.do_package_qa.5538
ERROR: Task (//chimera-src/meta-chimera/recipes-devtools/dotnet/dotnet_3.0.0.bb:do_package_qa) failed with exit code '1'

trampster, 2019-10-21:
This dotnet issue seem relevant https://github.com/dotnet/core/issues/1367 but Its from 2018 and I don't understand it well enough to draw any conclusions

srware, 2019-10-23:
This is true, the recipe has no support for ARM32. I can look into this though.

srware, 2019-10-23:
@trampster, I come across the same dependency issue as you building for arm which is obviously a build time issue with the .NET Core download. I think the only solution to this will be the recipe building .NET from source rather than using the binary distribution for each arch or hacking the curl recipe so it provides CURL_OPENSSL_3 instead of CURL_OPENSSL_4.

closed 2019-10-23 by srware

tbandtg, 2020-07-17:
What recipe that builds it from source?

srware, 2020-07-17:
@tbandtg , [meta-aspnet](https://github.com/Tragetaschen/meta-aspnet) built dotnet from source although I'm not sure how well maintained it is. I can re-evaluate binary support for arm32, it's been a while since I last checked. [This](https://github.com/RDunkley/meta-dotnet-core) layer seems to have had success in this regard although I have never tried it.

tbandtg, 2020-07-17:
It is arm64 but, I can not get it to work, I have also tried this one https://github.com/RDunkley/meta-dotnet-core but I get the following output 
ERROR: aspnet-core-3.1.3-r0 do_package_qa: QA Issue: /usr/share/dotnet/shared/Microsoft.NETCore.App/3.1.3/System.Net.Security.Native.so contained in package aspnet-core requires libgssapi_krb5.so.2()(64bit), but no providers found in RDEPENDS_aspnet-core? [file-rdeps] ERROR: aspnet-core-3.1.3-r0 do_package_qa: QA Issue: /usr/share/dotnet/shared/Microsoft.NETCore.App/3.1.3/System.Net.Http.Native.so contained in package aspnet-core requires libcurl.so.4()(64bit), but no providers found in RDEPENDS_aspnet-core? [file-rdeps] ERROR: aspnet-core-3.1.3-r0 do_package_qa: QA Issue: /usr/share/dotnet/shared/Microsoft.NETCore.App/3.1.3/System.Net.Http.Native.so contained in package aspnet-core requires libcurl.so.4(CURL_OPENSSL_3)(64bit), but no providers found in RDEPENDS_aspnet-core? [file-rdeps] ERROR: aspnet-core-3.1.3-r0 do_package_qa: QA Issue: /usr/share/dotnet/shared/Microsoft.NETCore.App/3.1.3/System.Net.Security.Native.so contained in package aspnet-core requires libgssapi_krb5.so.2(gssapi_krb5_2_MIT)(64bit), but no providers found in RDEPENDS_aspnet-core? [file-rdeps] ERROR: aspnet-core-3.1.3-r0 do_package_qa: QA run found fatal errors. Please consider fixing them. ERROR: Logfile of failure stored in: /home/agarza/var-fsl-yocto/build_wayland/tmp/work/aarch64-poky-linux/aspnet-core/3.1.3-r0/temp/log.do_package_qa.2754 ERROR: Task (/home/agarza/var-fsl-yocto/sources/meta-dotnet-core/recipes-runtime/aspnet-core/aspnet-core_3.1.3.bb:do_package_qa) failed with exit code '1'

srware, 2020-07-22:
@tbandtg , I have added a 5.0.0 preview recipe to master. I think this going to be the first .NET release where we can support arm on Yocto because the curl dependency has been completely removed. I've tried several workarounds with older versions to work around the curl dependency issue but there is no nice way of making it work. It all seems to stem from a broken curl package on Ubuntu which is what was used to compile the binary release of dotnet.

tbandtg, 2020-07-22:
Thanks much would there need to be any changes between master and the zeus branch that you know of?

srware, 2020-07-22:
Not that I can think of other than adding "zeus" to `LAYERSERIES_COMPAT` in `conf/layer.conf`