srware/meta-iot-cloud #69: Errors when building the azure package group

Bog999, 2020-04-13:
Hi, I'm trying to build the azure package group in branch 'thud' and the build fails while trying to fetch the git repositories.
I'm using Petalinux which is a based on yocto (thud) and is provided by xilinx. 
The build machine is a (virtual) Ubuntu 18.04.1. Git version is 2.17.1

Here is a log of the build.

`bogdan@bogdan-ubuntu18:/media/bogdan/Disk3/xilinx/build2$ bitbake packagegroup-cloud-azure
Parsing recipes: 100% |###########################################################################################################################| Time: 0:00:34
Parsing of 3258 .bb files complete (0 cached, 3258 parsed). 4383 targets, 170 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "1.40.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "arm-xilinx-linux-gnueabi"
MACHINE              = "zc702-zynq7"
DISTRO               = "petalinux"
DISTRO_VERSION       = "2019.2"
TUNE_FEATURES        = "arm armv7a vfp thumb neon callconvention-hard cortexa9"
TARGET_FPU           = "hard"
meta
meta-poky            = "HEAD:c7569a0469666b2886e57b22b912f83a6f610e79"
meta-perl
meta-python
meta-filesystems
meta-gnome
meta-multimedia
meta-networking
meta-webserver
meta-xfce
meta-initramfs
meta-oe              = "HEAD:638a176b1bf043bc71021233f37926b1a10aca12"
meta-browser         = "HEAD:d7e5fab5ec331edede71da7829d4b0d9bfc80ba4"
meta-qt5             = "HEAD:fb4ef4dd7bccfb53a5186c031105ea29320420ce"
meta-xilinx-bsp
meta-xilinx-pynq
meta-xilinx-contrib  = "HEAD:586e4001cdd90e3db67692b14bbe5daabc565478"
meta-xilinx-tools    = "HEAD:1f0161a321aac36b0dfe6d742f2019abd146d998"
meta-petalinux       = "HEAD:6f3c49f124d2ce1bfcea006091fd72063a63ed95"
meta-virtualization  = "HEAD:6743df6e68c8fa145a6472bac81a87b57f19f84d"
meta-openamp         = "HEAD:a1f9048de857908d94b2c48618decfe206176521"
meta-jupyter         = "HEAD:de267ef366fa9278e065506c4c4bd287803d0445"
meta-iot-cloud       = "thud:6efd11ba89523d2368dc6f85841cd3eba6bdb26a"

Initialising tasks: 100% |########################################################################################################################| Time: 0:00:01
Sstate summary: Wanted 625 Found 0 Missed 625 Current 375 (0% match, 37% complete)
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
ERROR: azure-umqtt-c-1.1.4-r0 do_unpack: Fetcher failure: Fetch command export PSEUDO_DISABLED=1; export DBUS_SESSION_BUS_ADDRESS="unix:path=/run/user/1000/bus"; export PATH="/media/bogdan/Disk3/xilinx/build2/tmp/sysroots-uninative/x86_64-linux/usr/bin:/media/bogdan/Disk3/xilinx/sources/core/scripts:/media/bogdan/Disk3/xilinx/build2/tmp/work/cortexa9t2hf-neon-xilinx-linux-gnueabi/azure-umqtt-c/1.1.4-r0/recipe-sysroot-native/usr/bin/arm-xilinx-linux-gnueabi:/media/bogdan/Disk3/xilinx/build2/tmp/work/cortexa9t2hf-neon-xilinx-linux-gnueabi/azure-umqtt-c/1.1.4-r0/recipe-sysroot/usr/bin/crossscripts:/media/bogdan/Disk3/xilinx/build2/tmp/work/cortexa9t2hf-neon-xilinx-linux-gnueabi/azure-umqtt-c/1.1.4-r0/recipe-sysroot-native/usr/sbin:/media/bogdan/Disk3/xilinx/build2/tmp/work/cortexa9t2hf-neon-xilinx-linux-gnueabi/azure-umqtt-c/1.1.4-r0/recipe-sysroot-native/usr/bin:/media/bogdan/Disk3/xilinx/build2/tmp/work/cortexa9t2hf-neon-xilinx-linux-gnueabi/azure-umqtt-c/1.1.4-r0/recipe-sysroot-native/sbin:/media/bogdan/Disk3/xilinx/build2/tmp/work/cortexa9t2hf-neon-xilinx-linux-gnueabi/azure-umqtt-c/1.1.4-r0/recipe-sysroot-native/bin:/media/bogdan/Disk3/xilinx/build2/tmp/sysroots-xsct/Vitis/2019.2/bin:/media/bogdan/Disk3/xilinx/build2/tmp/sysroots-xsct/Vitis/2019.2/gnu/microblaze/lin/bin:/media/bogdan/Disk3/xilinx/build2/tmp/sysroots-xsct/Vitis/2019.2/gnu/aarch32/lin/gcc-arm-none-eabi/bin:/media/bogdan/Disk3/xilinx/build2/tmp/sysroots-xsct/Vitis/2019.2/gnu/armr5/lin/gcc-arm-none-eabi/bin:/media/bogdan/Disk3/xilinx/build2/tmp/sysroots-xsct/Vitis/2019.2/gnu/aarch64/lin/aarch64-none/bin:/media/bogdan/Disk3/xilinx/sources/core/bitbake/bin:/media/bogdan/Disk3/xilinx/build2/tmp/hosttools"; export HOME="/home/bogdan"; git -c core.fsyncobjectfiles=0 submodule update --recursive --no-fetch failed with exit code 1, output:
Submodule path 'deps/c-utility': checked out '8290634e5c2d005643d5a7dd5f8e65ad7a4353c2'
Submodule path 'deps/c-utility/testtools/ctest': checked out 'a500106bb2339a988cf7f6445f2686333746efc4'
Submodule path 'deps/c-utility/testtools/testrunner': checked out '7ff63752ad696f8b641af96f8cb168db4a38b179'

warning: core.bare and core.worktree do not make sense
fatal: unable to set up work tree using invalid config
Unable to checkout 'a9be4df165633c0ac6fcff2bf94b1ae384856d39' in submodule path 'deps/c-utility/testtools/umock-c'
Failed to recurse into submodule path 'deps/c-utility'

ERROR: azure-umqtt-c-1.1.4-r0 do_unpack: Function failed: base_do_unpack
ERROR: Logfile of failure stored in: /media/bogdan/Disk3/xilinx/build2/tmp/work/cortexa9t2hf-neon-xilinx-linux-gnueabi/azure-umqtt-c/1.1.4-r0/temp/log.do_unpack.19193
ERROR: Task (/media/bogdan/Disk3/xilinx/sources/meta-iot-cloud/recipes-azure/azure-umqtt-c/azure-umqtt-c_1.1.4.bb:do_unpack) failed with exit code '1'
ERROR: azure-uamqp-c-1.2.4-r0 do_unpack: Fetcher failure: Fetch command export PSEUDO_DISABLED=1; export DBUS_SESSION_BUS_ADDRESS="unix:path=/run/user/1000/bus"; export PATH="/media/bogdan/Disk3/xilinx/build2/tmp/sysroots-uninative/x86_64-linux/usr/bin:/media/bogdan/Disk3/xilinx/sources/core/scripts:/media/bogdan/Disk3/xilinx/build2/tmp/work/cortexa9t2hf-neon-xilinx-linux-gnueabi/azure-uamqp-c/1.2.4-r0/recipe-sysroot-native/usr/bin/arm-xilinx-linux-gnueabi:/media/bogdan/Disk3/xilinx/build2/tmp/work/cortexa9t2hf-neon-xilinx-linux-gnueabi/azure-uamqp-c/1.2.4-r0/recipe-sysroot/usr/bin/crossscripts:/media/bogdan/Disk3/xilinx/build2/tmp/work/cortexa9t2hf-neon-xilinx-linux-gnueabi/azure-uamqp-c/1.2.4-r0/recipe-sysroot-native/usr/sbin:/media/bogdan/Disk3/xilinx/build2/tmp/work/cortexa9t2hf-neon-xilinx-linux-gnueabi/azure-uamqp-c/1.2.4-r0/recipe-sysroot-native/usr/bin:/media/bogdan/Disk3/xilinx/build2/tmp/work/cortexa9t2hf-neon-xilinx-linux-gnueabi/azure-uamqp-c/1.2.4-r0/recipe-sysroot-native/sbin:/media/bogdan/Disk3/xilinx/build2/tmp/work/cortexa9t2hf-neon-xilinx-linux-gnueabi/azure-uamqp-c/1.2.4-r0/recipe-sysroot-native/bin:/media/bogdan/Disk3/xilinx/build2/tmp/sysroots-xsct/Vitis/2019.2/bin:/media/bogdan/Disk3/xilinx/build2/tmp/sysroots-xsct/Vitis/2019.2/gnu/microblaze/lin/bin:/media/bogdan/Disk3/xilinx/build2/tmp/sysroots-xsct/Vitis/2019.2/gnu/aarch32/lin/gcc-arm-none-eabi/bin:/media/bogdan/Disk3/xilinx/build2/tmp/sysroots-xsct/Vitis/2019.2/gnu/armr5/lin/gcc-arm-none-eabi/bin:/media/bogdan/Disk3/xilinx/build2/tmp/sysroots-xsct/Vitis/2019.2/gnu/aarch64/lin/aarch64-none/bin:/media/bogdan/Disk3/xilinx/sources/core/bitbake/bin:/media/bogdan/Disk3/xilinx/build2/tmp/hosttools"; export HOME="/home/bogdan"; git -c core.fsyncobjectfiles=0 submodule update --recursive --no-fetch failed with exit code 1, output:
Submodule path 'deps/azure-c-shared-utility': checked out '8290634e5c2d005643d5a7dd5f8e65ad7a4353c2'
Submodule path 'deps/azure-c-shared-utility/testtools/ctest': checked out 'a500106bb2339a988cf7f6445f2686333746efc4'
Submodule path 'deps/azure-c-shared-utility/testtools/testrunner': checked out '7ff63752ad696f8b641af96f8cb168db4a38b179'
Submodule path 'deps/azure-c-testrunnerswitcher': checked out '7ff63752ad696f8b641af96f8cb168db4a38b179'
Submodule path 'deps/azure-ctest': checked out 'a500106bb2339a988cf7f6445f2686333746efc4'
Submodule path 'deps/umock-c': checked out 'a9be4df165633c0ac6fcff2bf94b1ae384856d39'
Submodule path 'deps/umock-c/deps/ctest': checked out 'a500106bb2339a988cf7f6445f2686333746efc4'
Submodule path 'deps/umock-c/deps/testrunner': checked out '7ff63752ad696f8b641af96f8cb168db4a38b179'

warning: core.bare and core.worktree do not make sense
fatal: unable to set up work tree using invalid config
Unable to checkout 'a9be4df165633c0ac6fcff2bf94b1ae384856d39' in submodule path 'deps/azure-c-shared-utility/testtools/umock-c'
Cloning into '/media/bogdan/Disk3/xilinx/build2/tmp/work/cortexa9t2hf-neon-xilinx-linux-gnueabi/azure-uamqp-c/1.2.4-r0/git/deps/umock-c/deps/ctest'...
done.
Failed to recurse into submodule path 'deps/azure-c-shared-utility'

ERROR: azure-uamqp-c-1.2.4-r0 do_unpack: Function failed: base_do_unpack
[... cut ...]
ERROR: Task (/media/bogdan/Disk3/xilinx/sources/meta-iot-cloud/recipes-azure/azure-uamqp-c/azure-uamqp-c_1.2.4.bb:do_unpack) failed with exit code '1'
NOTE: Tasks Summary: Attempted 1343 tasks of which 1251 didn't need to be rerun and 2 failed.
NOTE: Writing buildhistory

Summary: 2 tasks failed:
  /media/bogdan/Disk3/xilinx/sources/meta-iot-cloud/recipes-azure/azure-umqtt-c/azure-umqtt-c_1.1.4.bb:do_unpack
  /media/bogdan/Disk3/xilinx/sources/meta-iot-cloud/recipes-azure/azure-uamqp-c/azure-uamqp-c_1.2.4.bb:do_unpack
Summary: There were 4 ERROR messages shown, returning a non-zero exit code.`

I'd appreciate your help

srware, 2020-04-14:
Hi @Bog999,
There is something strange with the Azure git repos and their extensive use of submodules when fetching with bitbake. I've seen it quite often where there seems to be a lot of requests to the same github repos because they all share dependencies in the form of submodules which are fetched recursively and it seems to lead to some sort of timeout.

Things are slightly better with the recipes in master because I have been able to workaround having to fetch submodules for a number of the packages but for thud this isn't the case.

You can try building each of the following recipes one at a time so they are cached which generally helps:

azure-umqtt-c
azure-uamqp-c
azure-c-shared-utility
azure-iot-sdk-c

Bog999, 2020-04-15:
Thanks for the quick reply!
Unfortunately, I get the same error.
This is the log of the task that fails
```
DEBUG: Executing python function do_unpack
DEBUG: Executing python function base_do_unpack
DEBUG: Running export PSEUDO_DISABLED=1; export DBUS_SESSION_BUS_ADDRESS="unix:path=/run/user/1000/bus"; export SSH_AGENT_PID="1719"; export SSH_AUTH_SOCK="/run/user/1000/keyring/ssh"; export PATH="/media/bogdan/Disk3/xilinx/build2/tmp/sysroots-uninative/x86_64-linux/usr/bin:/media/bogdan/Disk3/xilinx/sources/core/scripts:/media/bogdan/Disk3/xilinx/build2/tmp/work/cortexa9t2hf-neon-xilinx-linux-gnueabi/azure-umqtt-c/1.1.4-r0/recipe-sysroot-native/usr/bin/arm-xilinx-linux-gnueabi:/media/bogdan/Disk3/xilinx/build2/tmp/work/cortexa9t2hf-neon-xilinx-linux-gnueabi/azure-umqtt-c/1.1.4-r0/recipe-sysroot/usr/bin/crossscripts:/media/bogdan/Disk3/xilinx/build2/tmp/work/cortexa9t2hf-neon-xilinx-linux-gnueabi/azure-umqtt-c/1.1.4-r0/recipe-sysroot-native/usr/sbin:/media/bogdan/Disk3/xilinx/build2/tmp/work/cortexa9t2hf-neon-xilinx-linux-gnueabi/azure-umqtt-c/1.1.4-r0/recipe-sysroot-native/usr/bin:/media/bogdan/Disk3/xilinx/build2/tmp/work/cortexa9t2hf-neon-xilinx-linux-gnueabi/azure-umqtt-c/1.1.4-r0/recipe-sysroot-native/sbin:/media/bogdan/Disk3/xilinx/build2/tmp/work/cortexa9t2hf-neon-xilinx-linux-gnueabi/azure-umqtt-c/1.1.4-r0/recipe-sysroot-native/bin:/media/bogdan/Disk3/xilinx/build2/tmp/sysroots-xsct/Vitis/2019.2/bin:/media/bogdan/Disk3/xilinx/build2/tmp/sysroots-xsct/Vitis/2019.2/gnu/microblaze/lin/bin:/media/bogdan/Disk3/xilinx/build2/tmp/sysroots-xsct/Vitis/2019.2/gnu/aarch32/lin/gcc-arm-none-eabi/bin:/media/bogdan/Disk3/xilinx/build2/tmp/sysroots-xsct/Vitis/2019.2/gnu/armr5/lin/gcc-arm-none-eabi/bin:/media/bogdan/Disk3/xilinx/build2/tmp/sysroots-xsct/Vitis/2019.2/gnu/aarch64/lin/aarch64-none/bin:/media/bogdan/Disk3/xilinx/sources/core/bitbake/bin:/media/bogdan/Disk3/xilinx/build2/tmp/hosttools"; export HOME="/home/bogdan"; git -c core.fsyncobjectfiles=0 branch --contains 870a2d5a496c38331e6bfaa4e0144c03fa65a6d7 --list master 2> /dev/null | wc -l
DEBUG: Running export PSEUDO_DISABLED=1; export DBUS_SESSION_BUS_ADDRESS="unix:path=/run/user/1000/bus"; export SSH_AGENT_PID="1719"; export SSH_AUTH_SOCK="/run/user/1000/keyring/ssh"; export PATH="/media/bogdan/Disk3/xilinx/build2/tmp/sysroots-uninative/x86_64-linux/usr/bin:/media/bogdan/Disk3/xilinx/sources/core/scripts:/media/bogdan/Disk3/xilinx/build2/tmp/work/cortexa9t2hf-neon-xilinx-linux-gnueabi/azure-umqtt-c/1.1.4-r0/recipe-sysroot-native/usr/bin/arm-xilinx-linux-gnueabi:/media/bogdan/Disk3/xilinx/build2/tmp/work/cortexa9t2hf-neon-xilinx-linux-gnueabi/azure-umqtt-c/1.1.4-r0/recipe-sysroot/usr/bin/crossscripts:/media/bogdan/Disk3/xilinx/build2/tmp/work/cortexa9t2hf-neon-xilinx-linux-gnueabi/azure-umqtt-c/1.1.4-r0/recipe-sysroot-native/usr/sbin:/media/bogdan/Disk3/xilinx/build2/tmp/work/cortexa9t2hf-neon-xilinx-linux-gnueabi/azure-umqtt-c/1.1.4-r0/recipe-sysroot-native/usr/bin:/media/bogdan/Disk3/xilinx/build2/tmp/work/cortexa9t2hf-neon-xilinx-linux-gnueabi/azure-umqtt-c/1.1.4-r0/recipe-sysroot-native/sbin:/media/bogdan/Disk3/xilinx/build2/tmp/work/cortexa9t2hf-neon-xilinx-linux-gnueabi/azure-umqtt-c/1.1.4-r0/recipe-sysroot-native/bin:/media/bogdan/Disk3/xilinx/build2/tmp/sysroots-xsct/Vitis/2019.2/bin:/media/bogdan/Disk3/xilinx/build2/tmp/sysroots-xsct/Vitis/2019.2/gnu/microblaze/lin/bin:/media/bogdan/Disk3/xilinx/build2/tmp/sysroots-xsct/Vitis/2019.2/gnu/aarch32/lin/gcc-arm-none-eabi/bin:/media/bogdan/Disk3/xilinx/build2/tmp/sysroots-xsct/Vitis/2019.2/gnu/armr5/lin/gcc-arm-none-eabi/bin:/media/bogdan/Disk3/xilinx/build2/tmp/sysroots-xsct/Vitis/2019.2/gnu/aarch64/lin/aarch64-none/bin:/media/bogdan/Disk3/xilinx/sources/core/bitbake/bin:/media/bogdan/Disk3/xilinx/build2/tmp/hosttools"; export HOME="/home/bogdan"; git -c core.fsyncobjectfiles=0 clone -s -n /media/bogdan/Disk3/poky/build/downloads/git2/github.com.Azure.azure-umqtt-c.git/ /media/bogdan/Disk3/xilinx/build2/tmp/work/cortexa9t2hf-neon-xilinx-linux-gnueabi/azure-umqtt-c/1.1.4-r0/git/
DEBUG: Running export PSEUDO_DISABLED=1; export DBUS_SESSION_BUS_ADDRESS="unix:path=/run/user/1000/bus"; export SSH_AGENT_PID="1719"; export SSH_AUTH_SOCK="/run/user/1000/keyring/ssh"; export PATH="/media/bogdan/Disk3/xilinx/build2/tmp/sysroots-uninative/x86_64-linux/usr/bin:/media/bogdan/Disk3/xilinx/sources/core/scripts:/media/bogdan/Disk3/xilinx/build2/tmp/work/cortexa9t2hf-neon-xilinx-linux-gnueabi/azure-umqtt-c/1.1.4-r0/recipe-sysroot-native/usr/bin/arm-xilinx-linux-gnueabi:/media/bogdan/Disk3/xilinx/build2/tmp/work/cortexa9t2hf-neon-xilinx-linux-gnueabi/azure-umqtt-c/1.1.4-r0/recipe-sysroot/usr/bin/crossscripts:/media/bogdan/Disk3/xilinx/build2/tmp/work/cortexa9t2hf-neon-xilinx-linux-gnueabi/azure-umqtt-c/1.1.4-r0/recipe-sysroot-native/usr/sbin:/media/bogdan/Disk3/xilinx/build2/tmp/work/cortexa9t2hf-neon-xilinx-linux-gnueabi/azure-umqtt-c/1.1.4-r0/recipe-sysroot-native/usr/bin:/media/bogdan/Disk3/xilinx/build2/tmp/work/cortexa9t2hf-neon-xilinx-linux-gnueabi/azure-umqtt-c/1.1.4-r0/recipe-sysroot-native/sbin:/media/bogdan/Disk3/xilinx/build2/tmp/work/cortexa9t2hf-neon-xilinx-linux-gnueabi/azure-umqtt-c/1.1.4-r0/recipe-sysroot-native/bin:/media/bogdan/Disk3/xilinx/build2/tmp/sysroots-xsct/Vitis/2019.2/bin:/media/bogdan/Disk3/xilinx/build2/tmp/sysroots-xsct/Vitis/2019.2/gnu/microblaze/lin/bin:/media/bogdan/Disk3/xilinx/build2/tmp/sysroots-xsct/Vitis/2019.2/gnu/aarch32/lin/gcc-arm-none-eabi/bin:/media/bogdan/Disk3/xilinx [... cut ...]
Submodule path 'deps/c-utility/testtools/testrunner': checked out '7ff63752ad696f8b641af96f8cb168db4a38b179'

warning: core.bare and core.worktree do not make sense
fatal: unable to set up work tree using invalid config
Unable to checkout 'a9be4df165633c0ac6fcff2bf94b1ae384856d39' in submodule path 'deps/c-utility/testtools/umock-c'
Failed to recurse into submodule path 'deps/c-utility'

DEBUG: Python function base_do_unpack finished
DEBUG: Python function do_unpack finished
ERROR: Function failed: base_do_unpack
```

Thanks

srware, 2020-04-16:
Hi @Bog999 ,

I have just pushed some changes to the thud branch which might help. If you can test and let me know it would be appreciated.

Thanks.

Bog999, 2020-04-16:
The build of azure-umqtt-c, azure-uamqp-c and azure-c-shared-utility succeeds but azure-iot-sdk-c fails
The log was too long to paste here [https://pastebin.com/gHCVEjk2](url)

scooter1556, 2020-04-16:
Thanks for testing.

I have done a few builds from scratch with the thud branch today and had no issues but I know this issue is quite random and is effected by many different factors. The ideal solution would be to remove the submodule fetching requirement for azure-iot-sdk-c but this isn't that easy. I will try and find some time to look into this some more and see if it will be possible.

srware, 2020-04-16:
Thanks for testing.

I have done a few builds from scratch with the thud branch today and had no issues but I know this issue is quite random and is effected by many different factors. The ideal solution would be to remove the submodule fetching requirement for azure-iot-sdk-c but this isn't that easy. I will try and find some time to look into this some more and see if it will be possible.

Bog999, 2020-04-16:
Thanks a lot!

srware, 2020-04-17:
I have pushed another change to the thud branch which in my tests allows the SDK recipe to build without fetching submodules. I'd appreciate if you can test.

I won't be able to fix the python-azure-iothub-X recipes due to the way the Python wrappers for the SDK are compiled so if you run into trouble with these and don't need them I would recommend removing 'python' from your 'packagegroup-cloud-azure' PACKAGECONFIG entry.

Bog999, 2020-04-18:
The build works now and I didn't have to disable the python recipes.
Thanks!

closed 2020-04-18 by Bog999