srware/meta-iot-cloud #70: Build cannot find azure iot libraries

bcis93, 2020-04-26:
Hi, I get the following errors when trying to build an application with azure-iot-sdk-c (using the thud branch)

```
| /home/bryan/aesd_project_build/build-openstlinuxweston-stm32mp1-aesd-project/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-ostl-linux-gnueabi/azure-app/1.0+gitAUTOINC+340686c3b0-r0/recipe-sysroot-native/usr/bin/arm-ostl-linux-gnueabi/../../libexec/arm-ostl-linux-gnueabi/gcc/arm-ostl-linux-gnueabi/8.2.0/ld: cannot find -luhttp
| /home/bryan/aesd_project_build/build-openstlinuxweston-stm32mp1-aesd-project/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-ostl-linux-gnueabi/azure-app/1.0+gitAUTOINC+340686c3b0-r0/recipe-sysroot-native/usr/bin/arm-ostl-linux-gnueabi/../../libexec/arm-ostl-linux-gnueabi/gcc/arm-ostl-linux-gnueabi/8.2.0/ld: cannot find -lprov_auth_client
| /home/bryan/aesd_project_build/build-openstlinuxweston-stm32mp1-aesd-project/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-ostl-linux-gnueabi/azure-app/1.0+gitAUTOINC+340686c3b0-r0/recipe-sysroot-native/usr/bin/arm-ostl-linux-gnueabi/../../libexec/arm-ostl-linux-gnueabi/gcc/arm-ostl-linux-gnueabi/8.2.0/ld: cannot find -lhsm_security_client
| collect2: error: ld returned 1 exit status
| ninja: build stopped: subcommand failed.
| WARNING: exit code 1 from a shell command.
| ERROR: Function failed: do_compile (log file is located at /home/bryan/aesd_project_build/build-openstlinuxweston-stm32mp1-aesd-project/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-ostl-linux-gnueabi/azure-app/1.0+gitAUTOINC+340686c3b0-r0/temp/log.do_compile.18252)

```

The recipe I am using to try to build my application is as follows:
```
LICENSE = "MIT"
LIC_FILES_CHKSUM = "file://${WORKDIR}/git/LICENSE;md5=96acd7c558849eaf93c9c522f1a2f334"

SRC_URI = "git://git@github.com/atharvanan1/aesd_linux_app.git;protocol=ssh;branch=master \
		   "

# Modify these as desired
PV = "1.0+git${SRCPV}"
SRCREV = "eb9f7e087da3c60c90b5fc911476c13f54ccbf1f"

S = "${WORKDIR}/git/azure"

DEPENDS_append += " \
	azure-iot-sdk-c \
	"

RDEPENDS_${PN} += " \
	iotedge-cli \
	iotedge-daemon \
	"

inherit cmake

do_configue () {
	cmake ..
}

do_install () {
	# Specify install commands here
	install -d ${D}${bindir_native}
	install -m 0755 ${S}/azure/azure_app ${D}${bindir_native}
}
```

Am I doing something wrong? Or are these libraries not getting installed correctly?

bcis93, 2020-04-27:
Never mind, I figured it out! For the missing uhttp library, I had to add a recipe to build it (and I opened a [pull request](https://github.com/intel-iot-devkit/meta-iot-cloud/pull/71) for this). For the other two libraries, I found that I needed to use the `use_edge_modules` flag while building azure-iot-sdk-c. I am planning on opening another pull request with this option added to the `PACKAGECONFIG`, if you would be open to that.

closed 2020-04-27 by bcis93

srware, 2020-04-27:
@bcis93 , I have just pushed some changes to master which will hopefully fix the issue. From looking at your PRs you came to the same conclusion as me about what was causing the problem.

You will need to specify the following in your config because edge modules are not built by default:
`PACKAGECONFIG_pn-azure-iot-sdk-c = "amqp edge http mqtt" `

bcis93, 2020-04-27:
@srware the fixes worked for me, thanks! As I mentioned in the original description, I am working on the thud branch, so I would like to get these changes there too, if you don't mind, and I would also like to update azure-iot-sdk-c to version 1.3.8. Let me know what you think - I'd be happy to submit a PR for this

srware, 2020-04-28:
@bcis93 , given thud is a stable branch I won't be able to update the SDK version other than for exceptional security related reasons. This is to keep compliance with Yocto maintenance guidelines.

I am happy to try and add support for compiling edge modules to the existing version though if this is relevant.

bcis93, 2020-04-28:
@srware that's ok then about the SDK update. But yes, adding support for compiling edge modules would be great!

srware, 2020-04-28:
@bcis93 , no problem. I will give you a heads up when it is done.