srware/meta-iot-cloud #73: Problem with node-red after enabling systemd

Bog999, 2020-05-10:
Hi, I'm trying to compile yocto on the "warrior" branch with meta-iot-cloud. After I added systemd to my image, the build started failing. I'll paste logs are below.

I'd appreciate your help.

'''
DEBUG: Executing shell function do_compile
npm WARN using --force I sure hope you know what you are doing.
npm ERR! Cannot read property 'replace' of null

npm ERR! A complete log of this run can be found in:
npm ERR!     /media/bogdan/Disk3/poky/build3/tmp/work/cortexa9t2hf-neon-poky-linux-gnueabi/node-red/1.0.2-r0/npm_cache/_logs/2020-05-10T10_13_05_397Z-debug.log
WARNING: /media/bogdan/Disk3/poky/build3/tmp/work/cortexa9t2hf-neon-poky-linux-gnueabi/node-red/1.0.2-r0/temp/run.do_compile.4316:1 exit 1 from 'npm --arch=arm --target_arch=arm --production --no-registry install'
ERROR: Execution of '/media/bogdan/Disk3/poky/build3/tmp/work/cortexa9t2hf-neon-poky-linux-gnueabi/node-red/1.0.2-r0/temp/run.do_compile.4316' failed with exit code 1:
npm WARN using --force I sure hope you know what you are doing.
npm ERR! Cannot read property 'replace' of null

npm ERR! A complete log of this run can be found in:
npm ERR!     /media/bogdan/Disk3/poky/build3/tmp/work/cortexa9t2hf-neon-poky-linux-gnueabi/node-red/1.0.2-r0/npm_cache/_logs/2020-05-10T10_13_05_397Z-debug.log
WARNING: /media/bogdan/Disk3/poky/build3/tmp/work/cortexa9t2hf-neon-poky-linux-gnueabi/node-red/1.0.2-r0/temp/run.do_compile.4316:1 exit 1 from 'npm --arch=arm --target_arch=arm --production --no-registry install'
'''

'''
0 info it worked if it ends with ok
1 verbose cli [ '/media/bogdan/Disk3/poky/build3/tmp/work/cortexa9t2hf-neon-poky-linux-gnueabi/node-red/1.0.2-r0/recipe-sysroot-native/usr/bin/node',
1 verbose cli   '/media/bogdan/Disk3/poky/build3/tmp/work/cortexa9t2hf-neon-poky-linux-gnueabi/node-red/1.0.2-r0/recipe-sysroot-native/usr/bin/npm',
1 verbose cli   '--arch=arm',
1 verbose cli   '--target_arch=arm',
1 verbose cli   '--production',
1 verbose cli   '--no-registry',
1 verbose cli   'install' ]
2 info using npm@6.9.0
3 info using node@v10.16.3
4 verbose npm-session ff84b6f41f74941e
5 silly install runPreinstallTopLevelLifecycles
6 silly preinstall node-red@1.0.2
7 info lifecycle node-red@1.0.2~preinstall: node-red@1.0.2
8 silly install loadCurrentTree
9 silly install readLocalPackageData
10 timing stage:loadCurrentTree Completed in 4279ms
11 silly install loadIdealTree
12 silly install cloneCurrentTreeToIdealTree
13 timing stage:loadIdealTree:cloneCurrentTree Completed in 2ms
14 silly install loadShrinkwrap
15 timing stage:rollbackFailedOptional Completed in 1ms
16 timing stage:runTopLevelLifecycles Completed in 4699ms
17 silly saveTree node-red@1.0.2
18 verbose stack TypeError: Cannot read property 'replace' of null
18 verbose stack     at quotemeta (/media/bogdan/Disk3/poky/build3/tmp/work/cortexa9t2hf-neon-poky-linux-gnueabi/node-red/1.0.2-r0/recipe-sysroot-native/usr/lib/node_modules/npm/lib/install/inflate-shrinkwrap.js:73:14)
18 verbose stack     at tarballToVersion (/media/bogdan/Disk3/poky/build3/tmp/work/cortexa9t2hf-neon-poky-linux-gnueabi/node-red/1.0.2-r0/recipe-sysroot-native/usr/lib/node_modules/npm/lib/install/inflate-shrinkwrap.js:77:20)
18 verbose stack     at inflatableChild (/media/bogdan/Disk3/poky/build3/tmp/work/cortexa9t2hf-neon-poky-linux-gnueabi/node-red/1.0.2-r0/recipe-sysroot-native/usr/lib/node_modules/npm/lib/install/inflate-shrinkwrap.js:99:22)
18 verbose stack     at BB.each (/media/bogdan/Disk3/poky/build3/tmp/work/cortexa9t2hf-neon-poky-linux-gnueabi/node-red/1.0.2-r0/recipe-sysroot-native/usr/lib/node_modules/npm/lib/install/inflate-shrinkwrap.js:55:12)
18 verbose stack     at tryCatcher (/media/bogdan/Disk3/poky/build3/tmp/work/cortexa9t2hf-neon-poky-linux-gnueabi/node-red/1.0.2-r0/recipe-sysroot-native/usr/lib/node_modules/npm/node_modules/bluebird/js/release/util.js:16:23)
18 verbose stack     at Object.gotValue (/media/bogdan/Disk3/poky/build3/tmp/work/cortexa9t2hf-neon-poky-linux-gnueabi/node-red/1.0.2-r0/recipe-sysroot-native/usr/lib/node_modules/npm/node_modules/bluebird/js/release/reduce.js:155:18)
18 verbose stack     at Object.gotAccum (/media/bogdan/Disk3/poky/build3/tmp/work/cortexa9t2hf-neon-poky-linux-gnueabi/node-red/1.0.2-r0/recipe-sysroot-native/usr/lib/node_modules/npm/node_modules/bluebird/js/release/reduce.js:144:25)
18 verbose stack     at Object.tryCatcher (/media/bogdan/Disk3/poky/build3/tmp/work/cortexa9t2hf-neon-poky-linux-gnueabi/node-red/1.0.2-r0/recipe-sysroot-native/usr/lib/node_modules/npm/node_modules/bluebird/js/release/util.js:16:23)
18 verbose stack     at Promise._settlePromiseFromHandler (/media/bogdan/Disk3/poky/build3/tmp/work/cortexa9t2hf-neon-poky-linux-gnueabi/node-red/1.0.2-r0/recipe-sysroot-native/usr/lib/node_modules/npm/node_modules/bluebird/js/release/promise.js:512:31)
18 verbose stack     at Promise._settlePromise (/media/bogdan/Disk3/poky/build3/tmp/work/cortexa9t2hf-neon-poky-linux-gnueabi/node-red/1.0.2-r0/recipe-sysroot-native/usr/lib/node_modules/npm/node_modules/bluebird/js/release/promise.js:569:18)
18 verbose stack     at Promise._settlePromiseCtx (/media/bogdan/Disk3/poky/build3/tmp/work/cortexa9t2hf-neon-poky-linux-gnueabi/node-red/1.0.2-r0/recipe-sysroot-native/usr/lib/node_modules/npm/node_modules/bluebird/js/release/promise.js:606:10)
18 verbose stack     at _drainQueueStep (/media/bogdan/Disk3/poky/build3/tmp/work/cortexa9t2hf-neon-poky-linux-gnueabi/node-red/1.0.2-r0/recipe-sysroot-native/usr/lib/node_modules/npm/node_modules/bluebird/js/release/async.js:142:12)
18 verbose stack     at _drainQueue (/media/bogdan/Disk3/poky/build3/tmp/work/cortexa9t2hf-neon-poky-linux-gnueabi/node-red/1.0.2-r0/recipe-sysroot-native/usr/lib/node_modules/npm/node_modules/bluebird/js/release/async.js:131:9)
18 verbose stack     at Async._drainQueues (/media/bogdan/Disk3/poky/build3/tmp/work/cortexa9t2hf-neon-poky-linux-gnueabi/node-red/1.0.2-r0/recipe-sysroot-native/usr/lib/node_modules/npm/node_modules/bluebird/js/release/async.js:147:5)
18 verbose stack     at Immediate.Async.drainQueues [as _onImmediate] (/media/bogdan/Disk3/poky/build3/tmp/work/cortexa9t2hf-neon-poky-linux-gnueabi/node-red/1.0.2-r0/recipe-sysroot-native/usr/lib/node_modules/npm/node_modules/bluebird/js/release/async.js:17:14)
18 verbose stack     at runCallback (timers.js:705:18)
19 verbose cwd /media/bogdan/Disk3/poky/build3/tmp/work/cortexa9t2hf-neon-poky-linux-gnueabi/node-red/1.0.2-r0/npmpkg
20 verbose Linux 5.0.0-31-generic
21 verbose argv "/media/bogdan/Disk3/poky/build3/tmp/work/cortexa9t2hf-neon-poky-linux-gnueabi/node-red/1.0.2-r0/recipe-sysroot-native/usr/bin/node" "/media/bogdan/Disk3/poky/build3/tmp/work/cortexa9t2hf-neon-poky-linux-gnueabi/node-red/1.0.2-r0/recipe-sysroot-native/usr/bin/npm" "--arch=arm" "--target_arch=arm" "--production" "--no-registry" "install"
22 verbose node v10.16.3
23 verbose npm  v6.9.0
24 error Cannot read property 'replace' of null
25 verbose exit [ 1, true ]
'''

srware, 2020-05-13:
Hi @Bog999 ,

I'm not able to replicate this on my side with the zeus branch. I'm not sure I understand why adding systemd to your build config would cause an npm error.

Have you tried doing a cleanall for the node-red package and rebuilding?

Bog999, 2020-05-13:
I tried to build the whole yocto image in a new directory, but I got the same result
Is it possible that the problem is in the host machine? I'm using Ubuntu 18.04.3. What else should I check?

srware, 2020-05-13:
@Bog999 , can you let me know what config was added which broke the node-red build?

Bog999, 2020-05-13:
I added the meta-updater layer and changed the distro to "poky-sota"
Then, I added these lines to the distro configuration
```
DISTRO_FEATURES_append = " systemd"
VIRTUAL-RUNTIME_init_manager = "systemd"
PREFERRED_RPROVIDER_virtual/network-configuration ??= "networkd-dhcp-conf"
DISTRO_FEATURES_BACKFILL_CONSIDERED += "sysvinit"
VIRTUAL-RUNTIME_initscripts = ""
```
The last step broke the build

srware, 2020-05-14:
@Bog999 , thanks for the config.

I have tried doing a default Poky image build with that config and again can't replicate the issue. I will try adding the meta-updater layer and changing the distro to "poky-sota" to see if this has an effect.

srware, 2020-05-15:
@Bog999 , I have tried doing a poky-sota build for Minnowboard and again everything completed without issue.

What MACHINE are you building for and are you pulling in any additional BSP layers for the hardware?

Bog999, 2020-05-15:
I'm using the [meta-xilinx](https://github.com/Xilinx/meta-xilinx) layers and the machine is zc702-zynq7

srware, 2020-05-15:
Hi @Bog999 , I have created a fresh build and pulled in the Xilinx layers, set machine to zc702-zynq7 and I am still not able to replicate the issue. Everything builds without issue on my end.

The original issue seems to be NPM related and generally caused by the package-lock.json but I would expect this to fail consistently.

I would suggest trying master/dunfell branches because Node recipes have been refactored to use new fetcher implementations but zeus isn't supported unfortunately.

The only thing I can suggest is changing your MACHINE to "qemuarm64" or "qemux86-64" and DISTRO to "poky", remove extraneous layers but keep the config which originally broke the build and try again.

What OS are you building on?

Bog999, 2020-05-17:
Building node-red succeeds when the MACHINE is set to "qemuarm" (32bit) but it fails when the MACHINE is "zc702-zynq7"

Bog999, 2020-05-18:
@srware Is it possible that you post here the configuration which you used in your tests? It's possible that I have misconfigured something.

srware, 2020-05-18:
@Bog999 , sure. I did another fresh build today with the following additional config on top of default "poky":

MACHINE = "zc702-zynq7"
DISTRO = "poky-sota"
PACKAGECONFIG_pn-boost = "python"
CORE_IMAGE_EXTRA_INSTALL += "node-red"
DISTRO_FEATURES_append = " systemd"
VIRTUAL-RUNTIME_init_manager = "systemd"
VIRTUAL-RUNTIME_initscripts = "systemd-compat-units"
DISTRO_FEATURES_BACKFILL_CONSIDERED += "sysvinit"

Layers:
BBLAYERS += "../meta-openembedded/meta-oe"
BBLAYERS += "../meta-openembedded/meta-python"
BBLAYERS += "../meta-openembedded/meta-networking"
BBLAYERS += "../meta-openembedded/meta-filesystems"
BBLAYERS += "../meta-iot-cloud"
BBLAYERS += "../meta-updater"
BBLAYERS += "../meta-xilinx/meta-xilinx-bsp"

Bog999, 2020-05-19:
I uninstalled npm from the host machine and did a clean. Now the build seems to be working.
Thanks for the help!

Should I close the issue?

srware, 2020-05-19:
@Bog999 , thanks for the update. Strange that something in your host configuration was messing with the bitbake build but glad that you resolved the issue.

I will close this now.

closed 2020-05-19 by srware