srware/meta-iot-cloud #78: ibm-iotf-java and aws-iot-device-sdk-java compilation fails on morty branch

tejas-volansys, 2020-08-29:
Hi, 

I am using meta-iot-cloud on the "morty" branch. Compilation of "ibm-iotf-java-0.2.6-r1" and "aws-iot-device-sdk-java_1.1.1" getting failed. It is trying to fetch the "nexus-staging-maven-plugin-1.6.3.pom" from "http://repo.maven.apache.org/maven2" but it is not longer accesible and suggesting to use "http://repo.maven.apache.org/maven2" instead.

Below is the log for the same

ERROR: ibm-iotf-java-0.2.6-r1 do_compile: Function failed: do_compile (log file is located at /home/tejas/mgw/fsl-release-bsp/build-x11/tmp/work/cortexa7hf-neon-poky-linux-gnueabi/ibm-iotf-java/0.2.6-r1/temp/log.do_compile.8456)
ERROR: Logfile of failure stored in: /home/tejas/mgw/fsl-release-bsp/build-x11/tmp/work/cortexa7hf-neon-poky-linux-gnueabi/ibm-iotf-java/0.2.6-r1/temp/log.do_compile.8456
Log data follows:
| DEBUG: Executing shell function do_compile
| [INFO] Scanning for projects...
| Downloading: http://repo.maven.apache.org/maven2/org/sonatype/plugins/nexus-staging-maven-plugin/1.6.3/nexus-staging-maven-plugin-1.6.3.pom
| 
| [ERROR] [ERROR] Some problems were encountered while processing the POMs:
| [ERROR] Unresolveable build extension: Plugin org.sonatype.plugins:nexus-staging-maven-plugin:1.6.3 or one of its dependencies could not be resolved: Failed to read artifact descriptor for org.sonatype.plugins:nexus-staging-maven-plugin:jar:1.6.3 @
|  @
| [ERROR] The build could not read 1 project -> [Help 1]
| [ERROR]
| [ERROR]   The project com.ibm.messaging:watson-iot:0.2.6 (/home/tejas/mgw/fsl-release-bsp/build-x11/tmp/work/cortexa7hf-neon-poky-linux-gnueabi/ibm-iotf-java/0.2.6-r1/git/pom.xml) has 1 error
| [ERROR]     Unresolveable build extension: Plugin org.sonatype.plugins:nexus-staging-maven-plugin:1.6.3 or one of its dependencies could not be resolved: Failed to read artifact descriptor for org.sonatype.plugins:nexus-staging-maven-plugin:jar:1.6.3: Could not transfer artifact org.sonatype.plugins:nexus-staging-maven-plugin:pom:1.6.3 from/to US (http://repo.maven.apache.org/maven2): Failed to transfer file: http://repo.maven.apache.org/maven2/org/sonatype/plugins/nexus-staging-maven-plugin/1.6.3/nexus-staging-maven-plugin-1.6.3.pom. Return code is: 501 , ReasonPhrase:HTTPS Required. -> [Help 2]
| [ERROR]
| [ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
| [ERROR] Re-run Maven using the -X switch to enable full debug logging.
| [ERROR]
| [ERROR] For more information about the errors and possible solutions, please read the following articles:
| [ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/ProjectBuildingException
| [ERROR] [Help 2] http://cwiki.apache.org/confluence/display/MAVEN/PluginManagerException
| WARNING: exit code 1 from a shell command.
| ERROR: Function failed: do_compile (log file is located at /home/tejas/mgw/fsl-release-bsp/build-x11/tmp/work/cortexa7hf-neon-poky-linux-gnueabi/ibm-iotf-java/0.2.6-r1/temp/log.do_compile.8456)
ERROR: Task (/home/tejas/mgw/fsl-release-bsp/sources/meta-iot-cloud/recipes-ibm/ibm-iotf-java/ibm-iotf-java_0.2.6.bb:do_compile) failed with exit code '1'
ERROR: aws-iot-device-sdk-java-1.1.1-r1 do_compile: Function failed: do_compile (log file is located at /home/tejas/mgw/fsl-release-bsp/build-x11/tmp/work/cortexa7hf-neon-poky-linux-gnueabi/aws-iot-device-sdk-java/1.1.1-r1/temp/log.do_compile.8455)
ERROR: Logfile of failure stored in: /home/tejas/mgw/fsl-release-bsp/build-x11/tmp/work/cortexa7hf-neon-poky-linux-gnueabi/aws-iot-device-sdk-java/1.1.1-r1/temp/log.do_compile.8455
Log data follows:
| DEBUG: Executing shell function do_compile
| [INFO] Scanning for projects...
| Downloading: http://repo.maven.apache.org/maven2/org/sonatype/plugins/nexus-staging-maven-plugin/1.6.3/nexus-staging-maven-plugin-1.6.3.pom
| 
| [ERROR] [ERROR] Some problems were encountered while processing the POMs:
| [ERROR] Unresolveable build extension: Plugin org.sonatype.plugins:nexus-staging-maven-plugin:1.6.3 or one of its dependencies could not be resolved: Failed to read artifact descriptor for org.sonatype.plugins:nexus-staging-maven-plugin:jar:1.6.3 @
|  @
| [ERROR] The build could not read 1 project -> [Help 1]
| [ERROR]
| [ERROR]   The project com.amazonaws:aws-iot-device-sdk-java-pom:1.1.1 (/home/tejas/mgw/fsl-release-bsp/build-x11/tmp/work/cortexa7hf-neon-poky-linux-gnueabi/aws-iot-device-sdk-java/1.1.1-r1/git/pom.xml) has 1 error
| [ERROR]     Unresolveable build extension: Plugin org.sonatype.plugins:nexus-staging-maven-plugin:1.6.3 or one of its dependencies could not be resolved: Failed to read artifact descriptor for org.sonatype.plugins:nexus-staging-maven-plugin:jar:1.6.3: Could not transfer artifact org.sonatype.plugins:nexus-staging-maven-plugin:pom:1.6.3 from/to US (http://repo.maven.apache.org/maven2): Failed to transfer file: http://repo.maven.apache.org/maven2/org/sonatype/plugins/nexus-staging-maven-plugin/1.6.3/nexus-staging-maven-plugin-1.6.3.pom. Return code is: 501 , ReasonPhrase:HTTPS Required. -> [Help 2]
| [ERROR]
| [ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
| [ERROR] Re-run Maven using the -X switch to enable full debug logging.
| [ERROR]
| [ERROR] For more information about the errors and possible solutions, please read the following articles:
| [ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/ProjectBuildingException
| [ERROR] [Help 2] http://cwiki.apache.org/confluence/display/MAVEN/PluginManagerException
| WARNING: exit code 1 from a shell command.
| ERROR: Function failed: do_compile (log file is located at /home/tejas/mgw/fsl-release-bsp/build-x11/tmp/work/cortexa7hf-neon-poky-linux-gnueabi/aws-iot-device-sdk-java/1.1.1-r1/temp/log.do_compile.8455)
ERROR: Task (/home/tejas/mgw/fsl-release-bsp/sources/meta-iot-cloud/recipes-aws/aws-iot-device-sdk-java/aws-iot-device-sdk-java_1.1.1.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 4114 tasks of which 4107 didn't need to be rerun and 2 failed.

Summary: 2 tasks failed:
  /home/tejas/mgw/fsl-release-bsp/sources/meta-iot-cloud/recipes-ibm/ibm-iotf-java/ibm-iotf-java_0.2.6.bb:do_compile
  /home/tejas/mgw/fsl-release-bsp/sources/meta-iot-cloud/recipes-aws/aws-iot-device-sdk-java/aws-iot-device-sdk-java_1.1.1.bb:do_compile
Summary: There was 1 WARNING message shown.
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.

scooter1556, 2020-09-07:
I am not able to replicate this unless I do not override the Maven mirrors to use http instead of https. With https I get errors pulling these packages for some reason which is why the Maven recipe provided by this layer is set to use http for that particular repository.

Given that the error you are receiving is stating that HTTPS is required I would assume that your build config or more likely network rules are not allowing plain HTTP requests and hence these packages are failing to fetch.

I will see if I can find time to understand why HTTPS fetching is not working when building with Yocto although I can't guarantee it will be priority given the recipes for these packages have been long since removed from the layer.

Thanks.

srware, 2020-09-07:
I am not able to replicate this unless I do not override the Maven mirrors to use http instead of https. With https I get errors pulling these packages likely due to TLSv1.1 support being removed from the central maven repo which is why the Maven recipe provided by this layer is set to use http for that particular repository.

Given that the error you are receiving is stating that HTTPS is required I would assume that your build config or more likely network rules are not allowing plain HTTP requests and hence these packages are failing to fetch.

I will see if I can find a fix for the TLS HTTPS fetching with older Java versions but given these recipes have been long since removed from the layer I can't guarantee it will be priority.

Thanks.

closed 2021-05-14 by srware