srware/meta-iot-cloud #88: Is it possible to get this updated to dotnet .300?

tbandtg, 2021-06-03:
Hello I tried to manually update it but I get a stat error on the install.

/aarch64-poky-linux/dotnet/5.0.300-r0/dotnet-5.0.300/host/fxr/gnu': No such file or directory
| WARNING: exit code 1 from a shell command.
| ERROR: Execution of '/mnt/nfs_share/edm_yocto2/build-xwayland-edm-g-imx8mp/tmp/work/aarch64-poky-linux/dotnet/5.0.300-r0/temp/run.do_install.11387' failed with exit code 1:
| cp: cannot stat '/mnt/nfs_share/edm_yocto2/build-xwayland-edm-g-imx8mp/tmp/work/aarch64-poky-linux/dotnet/5.0.300-r0/dotnet-5.0.300/host/fxr/gnu': No such file or directory
| WARNING: exit code 1 from a shell command.
| 
ERROR: Task (/mnt/nfs_share/edm_yocto2/sources/meta-iot-cloud/recipes-devtools/dotnet/dotnet_5.0.300.bb:do_install) failed with exit code '1'
NOTE: Tasks Summary: Attempted 7264 tasks of which 7255 didn't need to be rerun and 1 failed.

tbandtg, 2021-06-03:
Ill tell you why I want it, it is because 5.0 and 5.1 had a bug that they could not be properly debugged on an arm64, this was fixed in 5.2.

srware, 2021-06-03:
I will take a look at updating the recipe to 5.0.300.

srware, 2021-06-09:
Master should now be updated. Let me know if there are any issues.

Thanks.

closed 2021-06-09 by srware