srware/meta-iot-cloud #108: Upgrade s2n to 1.3.1?

rwmacleod, 2021-12-10:
Would you take a patch to uprev s2n to 1.3.1? 

We're seeing an intermittent build problem with 1.1.1  that may just be due to lack of memory in our container build but I'd just as soon upgrade before having someone debug the issue.

[136/515] TOPDIR/tmp-glibc/work/core2-64-wrs-linux/s2n/1.1.1-r0/recipe-sysroot-native/usr/bin/x86_64-wrs-linux/x86_64-wrs-linux-gcc -D_POSIX_C_SOURCE=200809L -Ds2n_EXPORTS -ITOPDIR/tmp-glibc/work/core2-64-wrs-linux/s2n/1.1.1-r0/git -ITOPDIR/tmp-glibc/work/core2-64-wrs-linux/s2n/1.1.1-r0/git/api -m64 -march=core2 -mtune=core2 -msse3 -mfpmath=sse -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed -fmacro-prefix-map=TOPDIR/tmp-glibc/work/core2-64-wrs-linux/s2n/1.1.1-r0=/usr/src/debug/s2n/1.1.1-r0                      -fdebug-prefix-map=TOPDIR/tmp-glibc/work/core2-64-wrs-linux/s2n/1.1.1-r0=/usr/src/debug/s2n/1.1.1-r0                      -fdebug-prefix-map=TOPDIR/tmp-glibc/work/core2-64-wrs-linux/s2n/1.1.1-r0/recipe-sysroot=                      -fdebug-prefix-map=TOPDIR/tmp-glibc/work/core2-64-wrs-linux/s2n/1.1.1-r0/recipe-sysroot-native=  -Wl,-z,relro,-z,now -fstack-protector-strong  -Og -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security  --sysroot=TOPDIR/tmp-glibc/work/core2-64-wrs-linux/s2n/1.1.1-r0/recipe-sysroot  -Og -g -feliminate-unused-debug-types -fmacro-prefix-map=TOPDIR/tmp-glibc/work/core2-64-wrs-linux/s2n/1.1.1-r0=/usr/src/debug/s2n/1.1.1-r0                      -fdebug-prefix-map=TOPDIR/tmp-glibc/work/core2-64-wrs-linux/s2n/1.1.1-r0=/usr/src/debug/s2n/1.1.1-r0                      -fdebug-prefix-map=TOPDIR/tmp-glibc/work/core2-64-wrs-linux/s2n/1.1.1-r0/recipe-sysroot=                      -fdebug-prefix-map=TOPDIR/tmp-glibc/work/core2-64-wrs-linux/s2n/1.1.1-r0/recipe-sysroot-native=  -pipe -Wno-error=array-parameter -Wno-error=discarded-qualifiers  -m64 -march=core2 -mtune=core2 -msse3 -mfpmath=sse -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed -fmacro-prefix-map=TOPDIR/tmp-glibc/work/core2-64-wrs-linux/s2n/1.1.1-r0=/usr/src/debug/s2n/1.1.1-r0                      -fdebug-prefix-map=TOPDIR/tmp-glibc/work/core2-64-wrs-linux/s2n/1.1.1-r0=/usr/src/debug/s2n/1.1.1-r0                      -fdebug-prefix-map=TOPDIR/tmp-glibc/work/core2-64-wrs-linux/s2n/1.1.1-r0/recipe-sysroot=                      -fdebug-prefix-map=TOPDIR/tmp-glibc/work/core2-64-wrs-linux/s2n/1.1.1-r0/recipe-sysroot-native=  -Wl,-z,relro,-z,now -fstack-protector-strong  -Og -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security  --sysroot=TOPDIR/tmp-glibc/work/core2-64-wrs-linux/s2n/1.1.1-r0/recipe-sysroot -fPIC -pedantic -std=gnu99 -Wall -Wimplicit -Wunused -Wcomment -Wchar-subscripts -Wuninitialized -Wshadow -Wcast-qual -Wcast-align -Wwrite-strings -Wno-deprecated-declarations -Wno-unknown-pragmas -Wformat-security -Wno-missing-braces -Wa,--noexecstack -Werror -fvisibility=hidden -DS2N_EXPORTS -DS2N_SIKE_P434_R3_ASM -DS2N_BIKE_R3_AVX2 -DS2N_BIKE_R3_AVX512 -DS2N_BIKE_R3_PCLMUL -DS2N_BIKE_R3_VPCLMUL -DS2N_KYBER512R3_AVX2_BMI2 -DS2N_ADX -DS2N_HAVE_EXECINFO -DS2N_CPUID_AVAILABLE -fPIC -DS2N_FALL_THROUGH_SUPPORTED -DS2N___RESTRICT__SUPPORTED -mavx2 -mavx -mbmi2 -MD -MT CMakeFiles/s2n.dir/pq-crypto/kyber_r3/KeccakP-1600-times4-SIMD256_avx2.c.o -MF CMakeFiles/s2n.dir/pq-crypto/kyber_r3/KeccakP-1600-times4-SIMD256_avx2.c.o.d -o CMakeFiles/s2n.dir/pq-crypto/kyber_r3/KeccakP-1600-times4-SIMD256_avx2.c.o -c TOPDIR/tmp-glibc/work/core2-64-wrs-linux/s2n/1.1.1-r0/git/pq-crypto/kyber_r3/KeccakP-1600-times4-SIMD256_avx2.c
ninja: build stopped: subcommand failed.

srware, 2021-12-10:
Hi @rwmacleod,

I will happily accept a PR for this however, upgrading s2n has a knock on for the rest of the AWS common runtime (CRT) libraries and associated packages (e.g. python-awcrt) which all need to be upgraded in situ so it may not be as trivial (or fast) as you want. Worth upgrading s2n in isolation to see if it helps your build problems. If it does I will gladly help get everything updated... it's already on the long TODO list.

hongxu-jia, 2021-12-28:
While DEBUG_BUILD = "1" in local.conf, `-Og' is added to cfalgs by
Yocto, it broke s2n build with `-Werror=stringop-overflow=' (enabled
by -Werror in s2n's CMakeLists.txt)

Here is the fix [1] to support gcc 11, I also submitted it to upstream [2]
[1] https://github.com/intel-iot-devkit/meta-iot-cloud/pull/110/
[2] https://github.com/aws/s2n-tls/pull/3160

srware, 2022-11-08:
@rwmacleod , I am likely to deprecate the AWS recipes from this repo given AWS now have an [https://github.com/aws4embeddedlinux/meta-aws](official meta layer) and I don't want to duplicate effort if their is an official supported option available.

If you haven't already and this is still an open are you able to test the AWS meta layer for your project and raise any issues encountered either here or on the layer repo?

closed 2023-01-19 by srware

rwmacleod, 2023-01-25:
Thanks for the info, we're switching layers so closing this issue is fine with me.