srware/meta-iot-cloud #115: Provide package.json to node-red recipe

Pan5ky, 2022-01-22:
Is it possible to add the original package.json used to generate the npm-shrinkwrap.json file to the node-red recipe? Currently, it is very cumbersome to install npm packages to node-red because there is no clean way to do so for a shrinkwrap file.

srware, 2022-01-23:
Hi @Pan5ky ,

I don't really understand the request. The package.json used to generate the npm-shrinkwrap.json is the same one which gets pulled down with the git source. The recently refactored openembedded NPM classes require this to maintain some kind of reproducible results.

> Currently, it is very cumbersome to install npm packages to node-red because there is no clean way to do so for a shrinkwrap file

What do you mean by this exactly? Any additional node-red packages (additional nodes etc...) should be handled in their own recipe.

Pan5ky, 2022-01-23:
Thank you @srware for the quick answer,

Let me explain what I am trying to do: I need to install some additional nodes in node-red, so I copied the complete recipe from this repository to my own layer. Then I edited the shrinkwrap file to include the nodes I need. This is not the best solution, of course, because shrinkwrap files are not meant to be used this way. A better solution would be to have a package.json, install all the necessary dependencies via npm, and then create the shrinkwrap file and use that in the recipe. Thats why I created this issue. I'd like to mention that I am not sure if this is the best way to add nodes to node-red and if there is a better way to do so, please show it to me.

srware, 2022-01-23:
OK, that's what I assumed you are trying to do.

Generally additional node-red nodes are contained in their own NPM package such as this: [IBM Watson nodes](https://www.npmjs.com/package/node-red-node-watson). The correct way to add them to your image would be to create a new recipe which depends on the node-red package, in this case the recipe would be called "node-red-node-watson". The node-red recipe could be used as a starting point for this. As long as the node packages are in the standard path on the target (e.g. /usr/lib/node_modules) node-red will pick them up. Trying to add them on top of the node-red package won't be a maintainable way forwards in reality.

Pan5ky, 2022-01-23:
Thank you! I found [this](https://www.yoctoproject.org/docs/current/mega-manual/mega-manual.html#creating-node-package-manager-npm-packages) documentation page which shows how to do exactly that (Section 7.22.7.2. Using the Registry Modules Method). This worked for me.

Super happy that I won't have to mess with the shrinkwrap file anymore haha.

closed 2022-01-23 by Pan5ky