srware/meta-iot-cloud #117: Update node-red version

amolina-chargevite, 2022-03-01:
Hi, I am using node-red in my yocto project image with the dunfell branch, but my problem is that the version of node-red installed with this branch is 1.0.2 and I need a more update version, for example the 2.2.2. Which files do I need to modify the version of node-red? I have tested with the json file but I am not familiar with json. Is there a command to download the last version of node-red? 
question:

srware, 2022-03-01:
Hi @amolina-chargevite ,

The master branch has a recipe for node-red 2.1.x but the master branch in it's entirety isn't compatible with dunfell due to the new override syntax being used (: instead of _).

You could try grabbing the latest [node-red recipe](https://github.com/intel-iot-devkit/meta-iot-cloud/tree/master/recipes-devtools/node-red) from the master branch and integrate them into your dunfell build. There are only a couple of syntax changes which will be required in the recipe file for it to work and these will be quickly flagged by bitbake when you begin to build.

amolina-chargevite, 2022-03-02:
Hi @srware,
I was trying copy the folder node-red of master branch to the dunfell branch but don't compile. Could you tell me what the changes are? Thanks in advance.

srware, 2022-03-03:
With latest dunfell the new override syntax is also supported so the only changes I needed to make was to add the following line to the node-red recipe:

`INSANE_SKIP_${PN} += "staticdev"`

closed 2022-03-23 by srware