srware/meta-iot-cloud #157: Update aws-iot-device-sdk-cpp-v2

Rafq77, 2022-10-14:
Is it possible to update the sdk to newer version? Currently is 1.12.6 at least 1.14.x would be preferred, because of the easier MqttClient handling and static ClientBootstrap instance. 

I wanted to do it myself but I don't understand the full scope of the `Build-static-and-shared-libs.patch`.

Rafq77, 2022-10-19:
Thank you for answering anyway :). I have downgraded my software to match the 1.12.x however it doesn't represent the quality that I'd prefer it would do. I consider my current solution as temporary until the meta layer would be updated. :) 

Otherwise, if I understood you correctly, I could try making my own recipe of the sdk with strictly static lib support. However it would be error prone while adjusting that patch to the newer version. 

If any update to the layer would come, would it be within this year or not earlier than 2023?

srware, 2022-10-25:
@Rafq77 , I should get some time this week to take a look at getting this updated.

There is also the [AWS official layer](https://github.com/aws4embeddedlinux/meta-aws) which is likely more up to date and might be worth a go.

Rafq77, 2022-11-02:
I also thought about that but then I was thinking that you did adjust sdk-cpp-v2 in your layer here. Therefore I have asked here first :)

srware, 2022-11-08:
@Rafq77 , I have tested the [https://github.com/aws4embeddedlinux/meta-aws](meta-aws) layer for all packages included in this layer and most of the issues I encountered previously have been fixed. I think it will be logical to deprecate the AWS recipes from this layer to prevent duplication and also make sure issues go back to AWS developers who have been assigned to support them.

Can you also test from your side and raise any issues you encounter using the official AWS repo?

Rafq77, 2022-11-10:
@srware Great to know, if there are now further dependencies I will try to change to meta-aws directly. This might take some time  though (few weeks). I'll aim for the report before Christmas.

Rafq77, 2022-12-23:
Hey @srware , It required a bigger rework on our side, but going staright for the kikstone branch in meta-aws solved the problem for us and helped align all the dependencies to one common revision. Thank You, Merry Christmas and Happy New Year :)

closed 2022-12-23 by Rafq77