srware/meta-iot-cloud #158: recipes-aws deleted with no corresponding recipes in meta-aws

ChenQi1989, 2022-11-14:
The recipes-aws is deleted by https://github.com/intel-iot-devkit/meta-iot-cloud/commit/3fc517e92f7ea939cf2635bccbaadf3e6f2995d2.
Yet I found that the following recipes lack corresponding recipes in meta-aws.

packagegroup-cloud-aws_0.17.bb
python3-awsiotsdk_1.5.12.bb
python3-awscrt_0.11.20.bb
python3-awscli_1.18.157.bb
aws-iot-device-sdk-cpp-v2-samples_1.12.6.bb
python3-gg-group-setup_0.5.3.bb

What's the plan for these recipes? Is the removal work done? Or is it still in progress?
Thanks in advance.

srware, 2022-11-14:
Hi @ChenQi1989 ,

This work isn't fully finished yet. I do have a plan for migration.

Let me go through each of the recipes and give details or meta-aws recipe names:

python3-awsiotsdk -> aws-iot-device-sdk-python-v2
python3-awscrt -> aws-crt-python
python3-awscli -> aws-cli
python3-gg-group-setup : This isn't an AWS project however, it is also unmaintained for over 2 years so needs to be removed ideally.
aws-iot-device-sdk-cpp-v2-samples : The samples package is missing from meta-aws. I will either add this back to this repo or contribute it to meta-aws which would be more logical.
packagegroup-cloud-aws : My plan would be to add any AWS recipes under dynamic layers so there is only a soft dependency on meta-aws. I can add the packagegroup here so that if meta-aws is present it can be used.

I don't like that the meta-aws layer doesn't follow naming conventions for python recipes but this needs to be raised as an issue on that repo.

Thoughts?

ChenQi1989, 2022-11-15:
Hi @srware ,
Thanks for your detailed explanation. It helps a lot.
I totally agree with your plan.
Once you finish the work, please do me a favor and let me know (maybe just close this issue at that time).
Thanks in advance :)

srware, 2023-01-17:
I have added packagegroup-cloud-aws back as a dynamic recipe if meta-aws is present in the build 6c8a8074bcc62607058ffce37abb5e33a40be87a

For the samples it turns out the SDK maintainers removed the root CMake build file for samples so they can now only be built individually. This is being discussed over at the meta-aws repo with the maintainers: https://github.com/aws4embeddedlinux/meta-aws/discussions/2153

ChenQi1989, 2023-01-19:
@srware Thanks for the update. 👍

closed 2023-01-19 by srware