srware/meta-iot-cloud #163: python3-setuptools-scm-git-archive-native fails to build in the kirkstone branch.

mabnhdev, 2023-01-10:
I get a build error in kirkstone when trying to build python3-setuptools-scm-git-archive-native.

I've attached a bash script that will reproduce the problem. [cheroot-repro.txt](https://github.com/intel-iot-devkit/meta-iot-cloud/files/10382349/cheroot-repro.txt)

I've also attached the full error log from Yocto. [log.do_compile.1391898.txt](https://github.com/intel-iot-devkit/meta-iot-cloud/files/10382509/log.do_compile.1391898.txt)

| /home/mabnhdev/cheroot-repro/poky/build/tmp/work/x86_64-linux/python3-setuptools-scm-git-archive-native/1.0-r0/recipe-sysroot-native/usr/lib/python3.10/site-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
|   warnings.warn(
| error: could not create 'build/bdist.linux-x86_64/wheel/setuptools_scm_git_archive/__init__.py': No such file or directory
| ERROR: 'python3 setup.py bdist_wheel ' execution failed.

scooter1556, 2023-01-11:
@mabnhdev this should fixed with commit 741ddf47429e03436191976511ce75b0c52d8c46

Thanks

srware, 2023-01-11:
@mabnhdev this should fixed with commit 741ddf47429e03436191976511ce75b0c52d8c46

Thanks

mabnhdev, 2023-01-12:
Thanks so much for the quick response.

Regards, Mike

closed 2023-01-12 by srware