srware/meta-iot-cloud #164: python3-jaraco-functools fails to build in Kirkstone

mabnhdev, 2023-01-12:
This component has as a dependency on python3-lang which is not defined in any of the required layers.

The command 'bitbake python3-jaraco-functools' fails to parse.

I've attached a bash script that will reproduce the problem.  [jaraco-repro.txt](https://github.com/intel-iot-devkit/meta-iot-cloud/files/10402466/jaraco-repro.txt)

I went looking around for python3-lang but I could only find a python(2) version in pypi.

----------------------------------------------------------

Loading cache: 100% |                                                                                               | ETA:  --:--:--
Loaded 0 entries from dependency cache.
Parsing recipes: 100% |##############################################################################################| Time: 0:00:28
Parsing of 2514 .bb files complete (0 cached, 2514 parsed). 3956 targets, 91 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies
ERROR: Nothing RPROVIDES 'python3-lang' (but /home/mabnhdev/tempora-repro/meta-iot-cloud/recipes-devtools/python/python3-jaraco-functools_1.20.bb RDEPENDS on or otherwise requires it)
NOTE: Runtime target 'python3-lang' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['python3-lang']
NOTE: Runtime target 'python3-jaraco-functools' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['python3-jaraco-functools', 'python3-lang']

ERROR: Required build target 'python3-tempora' has no buildable providers.
Missing or unbuildable dependency chain was: ['python3-tempora', 'python3-jaraco-functools', 'python3-lang']

Summary: There were 2 ERROR messages, returning a non-zero exit code.

srware, 2023-01-12:
This reminded me I had a bunch of patches to backport to Kirkstone. This should be fixed now.

srware, 2023-01-17:
Fixed with ec72cb74854366214c2746d468ef3a005bdc5852

closed 2023-01-17 by srware